realthunder/FreeCAD_assembly3 #641: [Assembly3] lofted faces sometimes do not place element coordinate system at center when added as an element

kevenwyld, 2021-05-29:
Please see video below.  When I create a lofted face and attempt to add it as an element, depending on the face's orientation the element coordinate system is at the corner of the face instead of the center.  I have not found a pattern exactly, but I'm able to reproduce relatively easily with the example I show here.  I've also attached my test file.

As a side note, if someone decides to open my test file you have to toggle "show element coordinate system" to be able to see them. This setting does not seem to persist, at least for me.

Video:

https://user-images.githubusercontent.com/11936535/120083687-b8e40000-c087-11eb-9229-14896027eb8e.mp4

Here's the FCStd:
[pyramid-and-block-test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6565278/pyramid-and-block-test.zip)

FreeCAD version information:
```
OS: Arch Linux (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.528.24301 +3686 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: d69f4add39382592e7b4eefbe5091953cda727b6
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```

kevenwyld, 2021-05-29:
Did some more digging.  This seems to only happen when a plane is trapezoidal.  Most other shapes seem have the ECS (if that's an OK acronym) centered correctly.

I obviously didn't test every shape in existence but all the ones I did test were predictable except for trapezoids.

https://user-images.githubusercontent.com/11936535/120086752-f0aa7200-c09e-11eb-9551-05478a51c675.mp4

luzpaz, 2022-04-08:
Can you still reproduce this on latest LinkDaily

kevenwyld, 2022-04-08:
So, yes and no.  But maybe it's as it should be now.  Here's a short clip of exactly the same file I had above but in today's release:

https://user-images.githubusercontent.com/11936535/162355702-4138c29b-faee-42a0-b830-187269d24148.mp4

As you can see the trapezoidal faces are centered correctly now.  It's worth noting that I didn't even recreate the elements so some of my old models will likely be broken from working around this issue (totally fine, I'm very happy it's fixed!).

The right triangle and crescent are still "wrong" though.  Or at least they are inconsistent with the others.  The center of the crescent is not on the crescent face at all, and the right triangle is on the long edge.

Let me know if you require any more testing or information, or if this is now intended behavior this can be closed.

Thanks for the excellent work as always!