realthunder/FreeCAD_assembly3 #645: Crash at load image

Dtays-meanderings, 2021-05-31:
Hi,

When using the new renderer and trying to load an image file, I receive an "Illegal storage access" error.

Log: *** Unhandled Exception!
Log:    ExpCode: 0xC0000005
Log:    ExpFlags: 0
Log:    ExpAddress: 0x21F379E0

OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git)
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

I love the new filled clipping plane. Thank you for all of your work!

snelweg, 2021-06-10:
Still crashes with the latest daily.

`OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.528.24301 +3686 (Git)
Build type: Release
Branch: LinkDaily
Hash: d69f4add39382592e7b4eefbe5091953cda727b6
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Dutch/Netherlands (nl_NL)
`

Dtays-meanderings, 2021-06-15:
I am without a computer for a spell. If someone can confirm the fix, I will close this.

Thank you for the speedy resolution Reallthunder!

-Danny

oficinerobotica, 2021-06-17:
@Dtays-meanderings 

It is solved for me with:
```
OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

[image]

closed 2021-06-17 by Dtays-meanderings