realthunder/FreeCAD_assembly3 #646: Treeview disable auto solid group does not work

Miksu80, 2021-06-01:
When activating auto solid group, solids are gathered correctly so that every solid is found under last active solid. When selecting disable auto solid group, treeview doesn't change at all.

[image]

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git)
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

realthunder, 2021-06-01:
This is by design. The user is free to ungroup any existing groups after enabling `Auto solid group`, so disabling this option won't affect any existing groups. It just stops auto grouping new features. You ungroup any features by right click that feature and choose `Ungroup solid features`.

Miksu80, 2021-06-01:
Designed to work this way, user-error detected :)

closed 2021-06-01 by Miksu80