realthunder/FreeCAD_assembly3 #647: [Feature request] Treeview folders for easier views/grouping

Miksu80, 2021-06-01:
It would be good to have possibility to create folders on treeview. Folders could have the hide/show -button next to it which would activate or deactivate views directly under it. This way you could clean the treeview and have easy hide/show on different assembly sections. There could be folders under folder and the visibility change would affect also hierarchically on every part under folder even if the part is on folder below main level.

realthunder, 2021-06-01:
Isn't this functionality already provided by the plain group? In asm3, there is also a group which is basically a specialization of the plain group.

[image]

Miksu80, 2021-06-01:
Is this possible to implement in right click -menu? I'd see it useful there since you might want to do it without touching the assemblies. I haven't checked assembly3 yet but its on my to-do-list..

realthunder, 2021-06-02:
The `non touching` behavior needs to be implemented by whatever assembly workbench you are using, and that is indeed one of the specialized behavior of the group in my asm3 workbench. asm3 group command has default shortcut `A, Z`. Just select multiple parts and press the shortcut to group them. You can assign shortcut to the core plain group command as well.

Miksu80, 2021-06-02:
Thanks, I'll close this :)

closed 2021-06-02 by Miksu80