realthunder/FreeCAD_assembly3 #649: Seg Fault with PartDesign fillet intersecting other geometry.

oficinerobotica, 2021-06-02:
I get a crash in certain situations when doing a fillet that will "intersect" other geometry.
In the example bellow, FC will crash as soon as the fillet value will force it to touch the adjacent case geometry. The refine property is set to true by default. 
[image]

`/tmp/.mount_FreeCA7E9Mue
FreeCAD 0.19, Libs: 2021.531R24301 +3702 (Git)
Assembly4 workbench loaded 
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f3617e6d210]
#1  0x7f3610133188 in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /tmp/.mount_FreeCA7E9Mue/usr/lib64/../lib/libTKBRep.so.7+0x28
#2  0x7f3610136818 in BRep_Tool::Curve(TopoDS_Edge const&, double&, double&) from /tmp/.mount_FreeCA7E9Mue/usr/lib64/../lib/libTKBRep.so.7+0x48
#3  0x7f3611b2fc0f in ChFi3d_Builder::PerformIntersectionAtEnd(int) from /tmp/.mount_FreeCA7E9Mue/usr/lib64/../lib/libTKFillet.so.7+0x38ef
#4  0x7f3611b37e1a in ChFi3d_Builder::PerformOneCorner(int, bool) from /tmp/.mount_FreeCA7E9Mue/usr/lib64/../lib/libTKFillet.so.7+0x10ba
#5  0x7f3611ad44fe in ChFi3d_Builder::PerformFilletOnVertex(int) from /tmp/.mount_FreeCA7E9Mue/usr/lib64/../lib/libTKFillet.so.7+0x51e
#6  0x7f3611ad54fc in ChFi3d_Builder::Compute() from /tmp/.mount_FreeCA7E9Mue/usr/lib64/../lib/libTKFillet.so.7+0xe5c
#7  0x7f3611c9a504 in BRepFilletAPI_MakeFillet::Build() from /tmp/.mount_FreeCA7E9Mue/usr/lib64/../lib/libTKFillet.so.7+0x34
#8  0x7f3610986b55 in BRepBuilderAPI_MakeShape::Shape() from /tmp/.mount_FreeCA7E9Mue/usr/lib64/../lib/libTKTopAlgo.so.7+0x45
#9  0x7f3612a3b199 in Part::TopoShape::makEShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /tmp/.mount_FreeCA7E9Mue/usr/lib64/Part.so+0x59
#10  0x7f3612a3b283 in Part::TopoShape::makEShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) from /tmp/.mount_FreeCA7E9Mue/usr/lib64/Part.so+0x83
#11  0x7f3612a3b3d9 in Part::TopoShape::makEFillet(Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, double, double, char const*) from /tmp/.mount_FreeCA7E9Mue/usr/lib64/Part.so+0x109
#12  0x7f3530195dde in PartDesign::Fillet::execute() from /tmp/.mount_FreeCA7E9Mue/usr/lib64/_PartDesign.so+0x20e
#13  0x7f361944b782 in App::DocumentObject::recompute() from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADApp.so+0xa2
#14  0x7f361283ef5a in Part::Feature::recompute() from /tmp/.mount_FreeCA7E9Mue/usr/lib64/Part.so+0xa
#15  0x7f353011549f in PartDesign::Feature::recompute() from /tmp/.mount_FreeCA7E9Mue/usr/lib64/_PartDesign.so+0xdf
#16  0x7f36193da509 in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADApp.so+0x209
#17  0x7f36194330e2 in App::Document::recomputeFeature(App::DocumentObject*, bool) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADApp.so+0x72
#18  0x7f34fa9d82df in PartDesignGui::TaskDressUpParameters::recompute() from /tmp/.mount_FreeCA7E9Mue/usr/lib64/PartDesignGui.so+0x1f
#19  0x7f3618387a18 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Core.so.5+0x6c0
#20  0x7f361a558754 in Gui::QuantitySpinBox::valueChanged(double) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADGui.so+0x44
#21  0x7f361a55a327 in Gui::QuantitySpinBox::updateFromCache(bool) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADGui.so+0xb7
#22  0x7f361a55af7f in Gui::QuantitySpinBox::stepBy(int) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADGui.so+0x14f
#23  0x7f3618c9dbcf in QAbstractSpinBox::keyPressEvent(QKeyEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Widgets.so.5+0x173
#24  0x7f361a55dfd4 in Gui::QuantitySpinBox::keyPressEvent(QKeyEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADGui.so+0xe4
#25  0x7f3618bc256d in QWidget::event(QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Widgets.so.5+0x1b3
#26  0x7f3618ca057f in QAbstractSpinBox::event(QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Widgets.so.5+0xbd
#27  0x7f3618b977f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#28  0x7f3618b9bcbb in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Widgets.so.5+0x467
#29  0x7f361a093eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADGui.so+0x89
#30  0x7f3618371606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Core.so.5+0x86
#31  /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Widgets.so.5(+0x1aff6c) [0x7f3618bd2f6c]
#32  0x7f3618b977f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#33  0x7f3618b9d31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#34  0x7f361a093eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADGui.so+0x89
#35  0x7f3618371606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Core.so.5+0x86
#36  0x7f3618713eaa in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Gui.so.5+0xca
#37  0x7f3618716939 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Gui.so.5+0x79
#38  0x7f3618703c65 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Gui.so.5+0x63
#39  /tmp/.mount_FreeCA7E9Mue/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5b010) [0x7f360f25c010]
#40  /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f3614a1423d]
#41  /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/./libglib-2.0.so.0(+0x554d1) [0x7f3614a144d1]
#42  /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f3614a14571]
#43  0x7f36183a00e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Core.so.5+0x5e
#44  0x7f361836e019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Core.so.5+0x187
#45  0x7f3618371e8f in QCoreApplication::exec() from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib/libQt5Core.so.5+0xfd
#46  0x7f361a00d92e in Gui::Application::runApplication() from /tmp/.mount_FreeCA7E9Mue/usr/bin/../lib64/libFreeCADGui.so+0x40e
#47  /tmp/.mount_FreeCA7E9Mue/usr/bin/FreeCADLink(+0x44be) [0x55c8792334be]
#48  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f3617e4e0b3]
#49  /tmp/.mount_FreeCA7E9Mue/usr/bin/FreeCADLink(+0x47d5) [0x55c8792337d5]`

Example file:
[BaseFeature.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6581759/BaseFeature.zip)

`OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
`

realthunder, 2021-06-04:
This is most likely a OCC problem, and is probably fixed in version after 7.5. Conda OCC release hasn't caught up with that version. Can you try using Windows release? It is bundled with a patched OCC 7.5.

oficinerobotica, 2021-06-04:
Indeed , testing with:
```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git)
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)```

FreeCAD won't crash anymore. When the fillet will touch the vertex common with both the case and the pad I will get only a "Command not done" error. 
What should I do with this issue? Close it?

realthunder, 2021-06-04:
Yeah. Windows release includes an OCC patched by one of our fellow forum member. And the patch has already been accepted by OpenCasCade if I am not wrong.

closed 2021-06-04 by realthunder