realthunder/FreeCAD_assembly3 #650: Pocket: Resulting shape is not a solid

krkpatrck, 2021-06-03:
ive been able to make pockets to bodies. All of a sudden i create a sketch and it will not pocket a simple hole into an active body?

realthunder, 2021-06-04:
Can you please post a screencast showing your steps?

realthunder, 2021-06-04:
And post a sample file showing the problem

herrgerd, 2021-06-06:
> 
> 
> ive been able to make pockets to bodies. All of a sudden i create a sketch and it will not pocket a simple hole into an active body?
> Using FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210528-glibc2.12-x86_64.AppImage

Have you tried `Reversed` when making the pocket? Seems like a dumb question, but sometimes it's the little things.

krkpatrck, 2021-06-12:
sorry i now think my model just broke somehow. i tried the same pocket in the freecad official and it does the same thing. disregard my comment

closed 2021-06-14 by oficinerobotica