realthunder/FreeCAD_assembly3 #652: "inconsistent constraints" when adding seemingly reasonable constraint

DodoDude700, 2021-06-05:
Hi there!

I'm trying to model a simple "test cube" using Assembly3, using eight corner pieces and twelve tube pieces. So far, I have gotten to the following, with four corners and three tubes added, and I'm now trying to add another tube. I have added the part and created a PlaneCoincident constraint (in green) to attach it to one of the corner pieces.

[image]

However, attempting to solve constraints with this constraint added results in an error (Failed to solve AlumatubeTest#Assembly: inconsistent constraints):

[image]

This does not seem to be an inconsistent or unreasonable constraint to add - the other tubes and corner pieces have been able to connect just fine using PlaneCoincident constraints, and I've even added some SameOrientation constraints to prevent the tubes from rotating (though I'm not sure if this is the correct way to do this, and removing them seems to make no difference as far as adding the fourth tube goes). If I remove the final constraint (PlaneCoincident006), the model solves just fine with the three tubes and four corners, but of course I need to add that to add the fourth tube.

I have attached my files as a ZIP, as the corner and tube are in separate FreeCAD files:

[Assembly3CubeTest.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6602232/Assembly3CubeTest.zip)

What's wrong with this model? If the constraint I am attempting to add is somehow incorrect, how can this be fixed?

Thanks!

oficinerobotica, 2021-06-05:
I did a quick test with the file and one thing I noticed is the 0 dof left reported by the solver if the mover tool is used on one of the components. Isn't the plane coincident leaving a rotational dof free?
@DodoDude700 
 If the last "Tube" is moved closer to the desired position, then a "plane coincident" constraint is applied, the later will move the tube correctly. Moving the component as close as possible to the desired position often helps the solver in correctly mating the parts.

realthunder, 2021-06-14:
PlaneConicdence is a very restrictive constraint. It removes 5 DOF, leaving just one rotational DOF. You'll almost always get redundant/inconsistent constraint problem when apply other constraint together with it on the same pair of parts. So obviously adding a SameOrientation constraint will get you in trouble. You can simply use the 'Lock angle' property of the PlaneCoincidence constraint to completely fix the parts without other constraints.

Or better, use Attachment constraint instead, which is a cheaper alternative to PlaneCoincidence that completely fixes two parts. If you get the orientation wrong, right click the constraint, the use flip element or manually change the 'Offset' property of one of the constraining element object to get what you want.