realthunder/FreeCAD_assembly3 #654: [Experimental Render] HiddenLine draw style flips face normal shading.

oficinerobotica, 2021-06-06:
If transparency is set to 0 and override face colors is checked in Edit->Preferences->Display-Draw styles-> Hidden Lines when switching to "HiddenLine" draw style,  the model faces will be displayed black (flipped normals?) . Toggling on/off "override face color" in preferences will display the model correctly. 
If using the clipping plane, toggling "invert fill color" will have the same effect. The model will be displayed black or with the correct color. This issue isn't present if the renderer is set to auto.
[image]

```
OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```

herrgerd, 2021-06-06:
I had a similar problem while editing sketches and using the `section view` command in sketcher.
When selecting, everything turns into a black square.
[image]

[image]

For me, this happens only with the Experimental renderer. Setting it to Auto there's no flaw here...

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git)
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

realthunder, 2021-06-10:
@oficinerobotica I couldn't reproduce the problem. Can you show me your hidden draw style settings, and the relevant model that having this problem?

@herrgerd These are probably two separate issues. Do you always get the all black when selecting or it just happen when you flip the sketch? I did notice this problem before as well, but it only happens when the sketch is flipped, i.e. edited from the bottom view. I am having hard time to trace out the cause of this problem, though. I suspect it could some Qt OpenGL problem at play.

oficinerobotica, 2021-06-10:
Attached the test file and link.user.cfg . Toggling "override face color" in the draw style preferences will display the model correctly .
[image]

[fcstd_and_config.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6628624/fcstd_and_config.zip)

oficinerobotica, 2021-06-15:
Tested with
```
OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```
and the issue isn't present any more.

closed 2021-06-15 by oficinerobotica