realthunder/FreeCAD_assembly3 #655: PartDesign SubShapeBinder sketch attachment to external geometry error.

oficinerobotica, 2021-06-07:
I really like the way new sketches can be attached but if: 
-clear selection -> create sketch tool in PartDesign
-select geometry on a different body
-the created subshape binder will give the error "<Exception> SubShapeBinder.cpp(1271): Failed to deduce bound geometry"

It works if the SubShapeBinder is created prior to the sketch then selected in the attachment dialog.
[image]

```
OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
```

realthunder, 2021-06-08:
I can't reproduce the problem. Can you please send me a project just before you create the sketch, and tell me which face you want to attach.

oficinerobotica, 2021-06-08:
In the attached file the active body is the cylinder. The selected face is on the box.
[image]

[boundGeometry.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6613707/boundGeometry.zip)

oficinerobotica, 2021-06-17:
Fixed in 
```
OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

[image]

closed 2021-06-17 by oficinerobotica