realthunder/FreeCAD_assembly3 #661: Issue: Tiny coordinate system symbol with new renderer

ngocminhdao88, 2021-06-10:
The coordinate symbol is so tiny with the new renderer.

With auto renderer:
[image]

With experiment renderer:
[image]

[example_file.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6632863/example_file.zip)

My FreeCAD info:
`OS: Ubuntu 20.04.2 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-06-14:
The problem has been recently fixed by realthunder/FreeCAD@3c93dd1d8c44f1138f8fb097c8aa6dd2995d41be

ngocminhdao88, 2021-06-16:
Update -> fixed with this FreeCAD version

> OS: Ubuntu 20.04.2 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

closed 2021-06-16 by ngocminhdao88