realthunder/FreeCAD_assembly3 #662: Issue: artifact with new renderer in clip view

ngocminhdao88, 2021-06-10:
I have some weird artifact in clip view with the new renderer

[image]

I am using the ArchDetail model from FreeCAD

My FreeCAD info:

> OS: Ubuntu 20.04.2 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-06-14:
If I am not mistaken, you are viewing the model with perspective camera, right? If so, the problem has been recently fixed by realthunder/FreeCAD@0f51061851e0e2583b25ffdf371a93537a76b6fb

ngocminhdao88, 2021-06-14:
Hi realthunder,

i am using the orthogonal view
[image]

Is there a way for me to save the clipped view for later use. Like the function SetWorkingPlaneProxy from Draft workbench?

realthunder, 2021-06-15:
Can you please post a sample file showing the problem?

realthunder, 2021-06-15:
The clip plane position will be remembered in next release. But you still need to manually enable the clip view.

ngocminhdao88, 2021-06-15:
Hi realthunder,

i am using the example "ArchDetail", which shipped with FreeCAD software. You can find it in the startup screen or in freecad_dir/data/examples/archdetail.fcstd

What i mean by "save for later use" is, that i have multiple clip view bookmarks with different offset or cut direction. With theses bookmarks i can quickly go back to the clipped view -> showing the customer the model from multiple angles, detail views for example.

realthunder, 2021-06-15:
I just checked the file. Section fill really only works for solids, while that Arch model mostly contains simple planes. So you probably want to turn off section filling. Use keyboard shortcut 'V, C' to bring up the clipping dialog. Uncheck 'Section fill' option.

I am not that familiar with Arch, but I see that Arch Workbench has a section plane feature. Maybe that's what you want?

ngocminhdao88, 2021-06-15:
i am also not familiar with Arch workbench :D, i just took the first file in startup page as example.

Is your fix already in the daily version 2021.531?

I have same issue with the EngineBlock model. The surface on top of Engine block solid causes weird artifact (BSpline001), although the cutting plane way bellow this surface. Hiding it seems solve the problem.

realthunder, 2021-06-15:
I just made a new release today, it does not have the fix yet. Like I said, the new `Clip section fill` feature only works with solid shape. I'll add auto detection of shape type in the future.

ngocminhdao88, 2021-06-20:
Close for now. Question is answered.

closed 2021-06-20 by ngocminhdao88