realthunder/FreeCAD_assembly3 #664: [Experimental Render] Line multiplier setting isn't respected.

oficinerobotica, 2021-06-12:
Using the experimental renderer and setting the "Line width multiplier" to a high value in Preferences->Display->Draw styles->Selection won't be correctly shown in the view port. Note that preview selection is shown multiplied but once selected the lines will be displayed the same thickness as  unselected lines. 

This feature is especially useful when in "Shadow" display mode as often, at certain angles, the shading of the model will make really difficult to distinguish the selected geometry. In the screenshot below note that there is also a face selected on the stepper motor. It is really hard to tell from that angle. A similar highlight feature would be useful for faces too. Or better of disabling shading for selected faces so they naturally pop up although I imagine this would be allot harder to implement.

[image]

thomasgi1, 2021-06-12:
The "Line Width Multiplier" seems to work only if the "Show selection on top" option is enabled too.

oficinerobotica, 2021-06-15:
@realthunder I was about to close this issue as solved but noticed your commit about "disable shadow on highlight rendering". In the screenshot below the highlighted face is selected but given the angle of the view combined with the shadow display mode the face  doesn't give any visual clue that is selected. This happens usually when a "light" color face is selected. 
If i click again on the face, the upper hierarchy geometry is selected. and the selected geometry is clear as day. It would be nice to have the same shading for individual selected faces.  In this case I use the experimental renderer but I get the same results if "Auto" is selected.

[image]

The main "Line multiplier" issue is solved.

```
OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

```

oficinerobotica, 2021-06-16:
Thank you 100:

closed 2021-06-16 by oficinerobotica