realthunder/FreeCAD_assembly3 #665: [Feature request] Mirror use preselect face or plane or ....

dgrabulosa, 2021-06-12:
When make a mirror, first select reference and then select mirror tool to make a mirror with selected reference.these enhancement  save some unnecessary clicks.
Thank you.

dgrabulosa, 2021-06-16:
Y test with simple model and it works.
Thank you for your attention to my suggestion.
Congratulations for your work.

closed 2021-06-16 by dgrabulosa

oficinerobotica, 2021-06-17:
Tested with the latest daily and now the PartDesign mirror automatically uses a preselected geometry as mirror plane.
```
OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```