realthunder/FreeCAD_assembly3 #667: Small problem with Asm3 SameOrientation constraint

jpg87Fr, 2021-06-13:
Hello, 

while I wanted to help a FreeCad forum member, I discovered a problem that does not occur on simple examples: SameOrientation constraint gives error message, while Perpendicular constraint works.
[Test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6644317/Test.zip)

realthunder, 2021-06-14:
There are three file inside the zip archive, but all of them contain only one part without constraint. Did you attach the wrong file?

jpg87Fr, 2021-06-14:
Sincerely sorry, I forgot the assembly file in my archive.
[Test2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6645839/Test2.zip)

closed 2021-06-14 by jpg87Fr

realthunder, 2021-06-14:
Did you close the issue by accident? Anyway, I checked your file. SameOrientation removes all three rotational DOF, while Perpendicular locks just one rotational DOF. So the error you get is from over constrain when using SameOrientation. For simple system, the solver can handle some redundancy, until it can't.

There is some peculiarity on how SolveSpace solver handles SameOrientation. The solver consider four cases of alignments to be the 'Same', where their axises is at 0, 90, 180 or 270 angles. I think that is why it might confuses user to think 'SameOrientation' is somewhat similar to Perpendicular. The critical differences here is that for SameOrientation, there are finite (discrete) number of solutions. The orientation of the first Part determines the orientation of the second, even though there might be multiple possibilities. While for Perpendicular, the second part can freely rotate along the two non-constrained direction.

jpg87Fr, 2021-06-14:
Thank you for this detailed explanation. I have used SameOrientation very often without problem, and I hastily deduced that only the normals to the designated planes were parallel.
From now on, I will be more careful.