realthunder/FreeCAD_assembly3 #668: Often crashes on clicking "Save"

StudioPetrikas, 2021-06-13:
Hi,
Having some real issues when saving files. It happens often enough for me to be afraid to click save, but not enough that I can reproduce consistently. It feels like it crashes more frequently when I'm messing around sketches, surfaces (Curves WB) and splines.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git)
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

Attaching the crash log, maybe there's something I can't figure out.
[crash.log](https://github.com/realthunder/FreeCAD_assembly3/files/6644322/crash.log)

Thank you in advance, loving and enjoying FreeCAD Link!

realthunder, 2021-06-14:
How did you get the crash log? Did you run it under Visual Studio? It would be better if you can capture the call stack while crash, instead of just the linked DLLs as your attachment.

StudioPetrikas, 2021-06-14:
> How did you get the crash log? Did you run it under Visual Studio?

The crashlog along with a crash dump file was generated in the users/freecad folder. I don't run FreeCAD with VS but I definitely need to learn how to - I will do that and return with the call stack. 
Thanks!

StudioPetrikas, 2021-06-14:
I remember I had an error pop up, something relating to "illegal storage access" or similar. I moved FreeCAD folder out of Program Files today, just to see whether it could be a weird Windows Administration problem. And it seems to have fixed the issue.

Sorry for raising an issue.

realthunder, 2021-06-15:
The `illegal storage access` indicates some serious unhandled exception somewhere in the code. I am not sure whether it is related to Admin problem.

StudioPetrikas, 2021-06-15:
It may have been unrelated, since most of the save-crashes were completely without error. 
Haven't crashed in 2 days now, where before it was every 2-3 saves. Thanks again.

oficinerobotica, 2021-07-05:
I experienced the SaveFile crash several times but it is not easy to find a clear pattern of when or why it happens. For this particular trace I'm not sure how to replicate the problem if needed. 

```
595.538975 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.670575 <asm3.sys> sys_slvs.py(68): dof remaining: 3
10.976821 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.484784 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.490162 <asm3.sys> sys_slvs.py(68): dof remaining: 3
1.755173 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.481616 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.491072 <asm3.sys> sys_slvs.py(68): dof remaining: 3
1.14444 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.478066 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.480584 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.709973 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.213541 <asm3.sys> sys_slvs.py(68): dof remaining: 3
0.260888 <asm3.sys> sys_slvs.py(68): dof remaining: 3
145.758292 <asm3.sys> sys_slvs.py(68): dof remaining: 0
13.931184 <asm3.sys> sys_slvs.py(68): dof remaining: 0
892.429175 <asm3.main> assembly.py(1128): _Element012 linked path [<App::Link object>, <PartDesign::Chamfer>]
0.052407 <asm3.main> assembly.py(1128): _Element013 linked path [<App::Link object>, <App::LinkElement object>]
118.816331 <asm3.sys> sys_slvs.py(68): dof remaining: 0
20.500543 <asm3.main> assembly.py(1128): _Element014 linked path [<App::Link object>, <PartDesign::Chamfer>]
0.053154 <asm3.main> assembly.py(1128): _Element015 linked path [<App::Link object>, <App::LinkElement object>]
20.572487 <asm3.sys> sys_slvs.py(68): dof remaining: 0
21.406435 <asm3.main> assembly.py(1128): _Element016 linked path [<App::Link object>, <PartDesign::Chamfer>]
0.055148 <asm3.main> assembly.py(1128): _Element017 linked path [<App::Link object>, <App::LinkElement object>]
21.593894 <asm3.sys> sys_slvs.py(68): dof remaining: 0
19.970225 <asm3.main> assembly.py(1128): _Element018 linked path [<App::Link object>, <PartDesign::Chamfer>]
0.056757 <asm3.main> assembly.py(1128): _Element019 linked path [<App::Link object>, <App::LinkElement object>]
20.665101 <asm3.sys> sys_slvs.py(68): dof remaining: 0
20.46494 <asm3.sys> sys_slvs.py(68): dof remaining: 0
7.097671 <asm3.sys> sys_slvs.py(68): dof remaining: 0
7.907616 <asm3.sys> sys_slvs.py(68): dof remaining: 0
6.842296 <asm3.sys> sys_slvs.py(68): dof remaining: 0
17.425068 <asm3.sys> sys_slvs.py(68): dof remaining: 0
4.306618 <asm3.sys> sys_slvs.py(68): dof remaining: 0
16.509653 <asm3.sys> sys_slvs.py(68): dof remaining: 0
5.972761 <asm3.sys> sys_slvs.py(68): dof remaining: 0
6.713136 <asm3.sys> sys_slvs.py(68): dof remaining: 0
108.393437 <asm3.sys> sys_slvs.py(68): dof remaining: 0
11.415749 <asm3.sys> sys_slvs.py(68): dof remaining: 0
<TopoShape> assembly.py(764)|TopoShapeEx.cpp(1049): hasher mismatch
<TopoShape> assembly.py(764)|TopoShapeEx.cpp(1049): hasher mismatch
<TopoShape> assembly.py(764)|TopoShapeEx.cpp(1049): hasher mismatch
<TopoShape> assembly.py(764)|TopoShapeEx.cpp(1049): hasher mismatch
107.10417 <asm3.sys> sys_slvs.py(68): dof remaining: 0
16.525865 <asm3.sys> sys_slvs.py(68): dof remaining: 0
0.792698 <asm3.sys> sys_slvs.py(68): dof remaining: 0
0.030162 <asm3.sys> sys_slvs.py(68): dof remaining: 0
0.182596 <asm3.sys> sys_slvs.py(68): dof remaining: 3
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f4f68fe5210]
#1  /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x67a1f8) [0x7f4f5e3271f8]
#2  /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x302f01) [0x7f4f5dfaff01]
#3  /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x30318e) [0x7f4f5dfb018e]
#4  /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x1487eb) [0x7f4f5ddf57eb]
#5  /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x39fef9) [0x7f4f5e04cef9]
#6  /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so(+0x3a02e2) [0x7f4f5e04d2e2]
#7  0x7f4f6b4b0b81 in SoFCVertexArrayIndexer::render(SoState*, cc_glglue const*, int, unsigned int, long const*, int const*, int) from /tmp/.mount_FreeCAjFK0ie/usr/bin/../lib64/libFreeCADGui.so+0x1d1
#8  0x7f4f6b4b399d in SoFCVertexCacheP::render(SoState*, SoFCVertexArrayIndexer*, int, long const*, int const*, int) from /tmp/.mount_FreeCAjFK0ie/usr/bin/../lib64/libFreeCADGui.so+0x40d
#9  0x7f4f6b4b3bb8 in SoFCVertexCache::renderLines(SoState*, int, int, bool) from /tmp/.mount_FreeCAjFK0ie/usr/bin/../lib64/libFreeCADGui.so+0x88
#10  0x7f4f6b4e14d3 in SoFCRendererP::renderOpaque(SoGLRenderAction*, std::vector<DrawEntry, std::allocator<DrawEntry> >&, std::vector<unsigned long, std::allocator<unsigned long> >&, int) from /tmp/.mount_FreeCAjFK0ie/usr/bin/../lib64/libFreeCADGui.so+0x453
#11  0x7f4f6b4e2578 in SoFCRenderer::render(SoGLRenderAction*) from /tmp/.mount_FreeCAjFK0ie/usr/bin/../lib64/libFreeCADGui.so+0x668
#12  0x7f4f6b4d3f87 in SoFCRenderCacheManager::render(SoGLRenderAction*) from /tmp/.mount_FreeCAjFK0ie/usr/bin/../lib64/libFreeCADGui.so+0x1d7
#13  0x7f4f6b5175ad in Gui::SoFCUnifiedSelection::GLRenderBelowPath(SoGLRenderAction*) from /tmp/.mount_FreeCAjFK0ie/usr/bin/../lib64/libFreeCADGui.so+0x15d
#14  0x7f4f68ba4245 in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /tmp/.mount_FreeCAjFK0ie/usr/bin/../lib64/../lib/libCoin.so.80+0x3e5
#15  0x7f4f68979149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAjFK0ie/usr/bin/../lib64/../lib/libCoin.so.80+0x69
[... cut ...]
#88  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f4f68fc60b3]
#89  /tmp/.mount_FreeCAjFK0ie/usr/bin/FreeCADLink(+0x47d5) [0x55daa2ea47d5]
```

```
OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

bennvan, 2021-07-15:
I don't know if it's from the same cause but I can reproduce a crash just by renaming a body and immediately saving after. This only happens when the experimental renderer is enabled. 

[image]

OS: Ubuntu 20.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/Australia (en_AU)

luzpaz, 2022-01-12:
@StudioPetrikas please retest on latest and report back. Thanks!

StudioPetrikas, 2022-01-12:
> @StudioPetrikas please retest on latest and report back. Thanks!

Haven't had a crash since my last post here. Very stable here.

luzpaz, 2022-01-12:
Thanks. Closing ticket. If re-appears, please re-open.

closed 2022-01-12 by luzpaz