realthunder/FreeCAD_assembly3 #670: DXF file can not be imported

ceremcem, 2021-06-14:
> This is possibly an `upstream` bug, but I'd like to open an issue here anyway. 

# Description 

`Import`ing a specific DXF gives the following exception: 

```
DXF libraries need to be updated. Trying to download...
downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfColorMap.py ...
downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfImportObjects.py ...
downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfLibrary.py ...
downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfReader.py ...
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/aea/fc-build/Release/Mod/Draft/importDXF.py", line 2824, in insert
    getDXFlibs()
  File "/home/aea/fc-build/Release/Mod/Draft/importDXF.py", line 202, in getDXFlibs
    import dxfColorMap, dxfLibrary, dxfReader
<class 'SyntaxError'>: invalid syntax (dxfReader.py, line 88)
```

Is there anyone having a workaround for this? 

# Attachments 

Contents of the DXF file: 
[image]

DXF File: 
[RT-507_dxf.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6649778/RT-507_dxf.zip)

# Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27873 (Git)
Build type: Release
Branch: LinkStage3
Hash: 1578dc89cb181b96e75d9fb128d82b0ec1460291
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2021-06-15:
This DXF import looks quite old. According to its [readme](https://github.com/yorikvanhavre/Draft-dxf-importer#readme), it is not need unless you explicitly want to use the legacy importer. Any reason for doing that? I tried the built-in importer. Seems working fine.

ceremcem, 2021-06-15:
What is "built-in importer"? First I tried "File->Import", now I tried "File->Open", both are giving the same exception. Then I uninstalled `dxf-library` and restarted FreeCAD and "File->Open" the dxf file, here is the exception: 

```
06:40:41  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/aea/fc-build/Release/Mod/Draft/importDXF.py", line 2780, in open
    processdxf(doc, filename)
  File "/home/aea/fc-build/Release/Mod/Draft/importDXF.py", line 2191, in processdxf
    drawing = dxfReader.readDXF(filename)
  File "/home/aea/.FreeCAD/Macro/dxfReader.py", line 372, in readDXF
    drawing = sm.run(infile)
  File "/home/aea/.FreeCAD/Macro/dxfReader.py", line 95, in run
    (newState, cargo) = handler(cargo)
  File "/home/aea/.FreeCAD/Macro/dxfReader.py", line 299, in start_section
    obj = handleBlock(obj, infile) # we need to find all there contents
  File "/home/aea/.FreeCAD/Macro/dxfReader.py", line 239, in handleBlock
    obj = handleObject(infile)
  File "/home/aea/.FreeCAD/Macro/dxfReader.py", line 196, in handleObject
    line = infile.readline()
  File "/usr/lib/python3.7/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
<class 'UnicodeDecodeError'>: 'utf-8' codec can't decode byte 0xdd in position 7362: invalid continuation byte
```

How should I import that file?

### Note

Opening that DXF with LibreCAD and saving as "DXF 2007" format works fine after uninstalling `dxf-library`.

realthunder, 2021-06-15:
Uncheck the legacy options in the preference.

[image]

ceremcem, 2021-06-15:
Okay, when I uncheck that checkbox, FreeCAD imports/opens the original DXF file. 

However: 

1. The result is not correct for some bits: 

[image]

2. When I uncheck the "Use legacy python importer" option, the DXF 2007 format can not be imported: 
```
07:01:20  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/aea/fc-build/Release/Mod/Draft/importDXF.py", line 2795, in open
    Import.readDXF(filename)
<class 'RuntimeError'>: gp_Trsf::SetValues, null determinant
```

# Attachments 

[pcb.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6653510/pcb.zip)

realthunder, 2021-06-15:
Just pushed a fix in both of my branches. You can sync and try.

ceremcem, 2021-06-15:
Works as expected. Thanks.

closed 2021-06-15 by ceremcem