realthunder/FreeCAD_assembly3 #671: AppImage lib/libstdc++.so.6 is missing

astralaster, 2021-06-15:
Hi, I tried to run the current AppImage release on arch linux and it gave me this output:
`
/tmp/.mount_FreeCAOJHiep
FreeCAD 0.19, Libs: 2021.528R24301 +3572 (Git)
Assembly4 workbench loaded
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAOJHiep/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAOJHiep/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAOJHiep/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCAOJHiep/AppRun: line 46:  7350 Aborted                 (core dumped) "$PREFIX/bin/$name" "$@"
./FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210528-glibc2.12-x86_64.AppImage
`
I think libstdc++ is not included in the AppImage?

luzpaz, 2021-06-16:
@looooo I can confirm that the appimage doesn't work. Can you check when you get a moment?

astralaster, 2021-06-17:
I did a quick flatpak image of the current source:
https://drive.google.com/file/d/1N0nFCO1sALGdDeOSp0of83A1U0sy-Xld/view?usp=sharing
To install just download it and type: flatpak install org.freecadweb.FreeCAD.flatpak
I will publish my buildprocess in the near future. Maybe it would be possible to publish this version of freecad on flatpak official, too? :)

realthunder, 2021-06-17:
That's great! Wait for your build instruction. I am not familiar with flatpak. When I try to install it, it fails with the message `requires the runtime org.kde.Platform/x86_64/5.15 which was not found`. Any command option to install dependencies?

astralaster, 2021-06-17:
Yes, I ran into the same "problem". I dont know why flatpak does not automatically install the needed runtime. But you can do it manually before the install:

```
flatpak install org.kde.Platform
Looking for matches…
Found similar ref(s) for ‘org.kde.Platform’ in remote ‘flathub’ (system).
Use this remote? [Y/n]:
Similar refs found for ‘org.kde.Platform’ in remote ‘flathub’ (system):

   1) runtime/org.kde.Platform/x86_64/5.10
   2) runtime/org.kde.Platform/x86_64/5.12
   3) runtime/org.kde.Platform/x86_64/5.11
   4) runtime/org.kde.Platform/x86_64/5.13
   5) runtime/org.kde.Platform/x86_64/5.14
   6) runtime/org.kde.Platform/x86_64/5.15
   7) runtime/org.kde.Platform/x86_64/5.9

Which do you want to use (0 to abort)? [0-7]:
```

johnsonm, 2021-06-20:
I'm seeing the same problem on Fedora 34. It worked on Fedora 33 because the system library was compatible, but it's moved forward. libstdc++ compatibility has always been a problem, so including it in the appimage would make it compatible with a wider range of systems, not just Arch Linux. ☺

astralaster, 2021-06-21:
You can now check out my build process at my fork of the official freecad flathub build repository:
https://github.com/astralaster/org.freecadweb.FreeCAD

Turiok, 2021-06-26:
@johnsonm  I have a Manjaro but the same problem.
The bad library libstdc++ is in the /tmp.mount_Freecad_XXXX/usr/lib/
But I can't copy or whatever with this folder with the version in my system.

I will try the flatpak too.

Turiok, 2021-06-26:
@astralaster The pre-release launch Freecad for me.
But I'm not sure if it's the realthunder's version.
I don't know flatpak. But the choice for installation doesn't mention realthunder.
The branch linkstage3 doesn't exist on the realthunder repo.
And the splashscreen is not from realthunder too.

astralaster, 2021-06-26:
@Turiok
Please check the about info. There should be a info about the linkstage3 branch.

Turiok, 2021-06-27:
[image]
I don't see the linkstage3 branch.

I think you should add in the readme how to get the Desktop Icon or how to launch Freecad. 
And add a chapter for the copy or the AppImage configuration in Flatpak folder tree.
And maybe change the name of the installation to see realthunder version.

I will prefer waiting with a corrected version because usually I get the assembly 3 version not just the Freecad version.
Thanks for your work.

astralaster, 2021-06-27:
Please check how you did the installation. I did download the pre-release and installed it. The about dialog shows:
OS: KDE Flatpak runtime (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.14555 (Git shallow)
Build type: Release
Branch: LinkStage3
Hash: 7437caf7c3e4c6fdcb9f29b6c05596d1268c5248
Python version: 3.8.8
Qt version: 5.15.3
Coin version: 4.0.0
OCC version: 7.3.0
Locale: English/United States (en_US)

Turiok, 2021-06-27:
@astralaster  It works!! :)
My bad, I installed your version once. Then uninstall it. And with the reinstallation I write "flatpak install org.freecadweb.FreeCAD" instead your file. Now I have the linkstage3 branch. 

I can continue my learning to freeCAD. Thx :)

Turiok, 2021-07-07:
Related to #687

Turiok, 2021-07-07:
The workaround in the last link of the other issue works for me.
with the command : 
`LD_PRELOAD=/usr/lib/libstdc++.so /opt/appimages/Freecad.AppImage`

johnsonm, 2021-07-07:
Thanks @Turiok !

For me on Fedora 34, it's a slightly different command line:
```
LD_PRELOAD=/lib64/libstdc++.so.6 FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210615-glibc2.12-x86_64.AppImage
```

johnsonm, 2021-07-08:
Thanks @Turiok !

For me on Fedora 34, it's a slightly different command line, e.g.:
```
LD_PRELOAD=/lib64/libstdc++.so.6 FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210615-glibc2.12-x86_64.AppImage
```

luzpaz, 2021-07-24:
@realthunder can you make a new release for all appimages now that upstream conda has fixed libstd++ issue?  
https://github.com/FreeCAD/FreeCAD-Bundle/issues/73#issuecomment-885390140

realthunder, 2021-07-25:
Anyone tried the latest release? Is it fixed? I checked the libstdc++ inside the AppImage, it is no longer linked to GLIBCXX_3.4.29.

luzpaz, 2021-07-25:
@realthunder I'm using the Daily and I still get this:  
```
$ ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210717-glibc2.12-x86_64_5c4e1b1219f005c0cddc3f87c889f98e.AppImage 
/tmp/.mount_FreeCAaEAomu
FreeCAD 0.19, Libs: 2021.717R24301 +3842 (Git)
Assembly4 workbench loaded 
Sheet Metal workbench loaded
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAaEAomu/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAaEAomu/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCAaEAomu/AppRun: line 46: 10446 Aborted                 (core dumped) "$PREFIX/bin/$name" "$@"
```

johnsonm, 2021-07-31:
Likewise on Fedora34 I get the same error from FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage and still need to run it with `LD_PRELOAD=/lib64/libstdc++.so.6`

luzpaz, 2021-07-31:
@realthunder I don't see the problem anymore on the appimage of Master (loooo triggered a rebuild and the updated libstdc++ were used and fixed the bug).

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.25288 (Git) AppImage
Build type: Release
Branch: master
Hash: bbb557aab6129efd9486d6df19f06c2976fdb7e5
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)
```

ademlabs, 2021-08-12:
Running the Daily or Stable appimage still produces this error. Doing an LD_PRELOAD to the c++ library solved my issue for now:

`$ LD_PRELOAD=/usr/lib/libstdc++.so ./FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage`

luzpaz, 2021-09-12:
Should be fixed in the 9.11 build: https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11 

Closing ticket. Please re-open if persists.

closed 2021-09-12 by luzpaz

johnsonm, 2021-09-14:
There are no AppImages visible to me in the 9.11 / 0.11 release page. @luzpaz am I missing something?

realthunder, 2021-09-14:
09.11 release has a serious bug, So I have taken off the images. I'll make a new release soon.

johnsonm, 2021-09-16:
I confirm that FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210916-glibc2.12-x86_64.AppImage resolves the problem on Fedora 34. Thanks!