realthunder/FreeCAD_assembly3 #673: shape does not change when sketch is changed in groove

mwlandolsi, 2021-06-16:
Hello,
When I create a groove function with a sketch, the operation is done ok. but when I change the sketch and even click on rebuild, the shape does not change.
this example shows the change of the sketch support surface offset
[image]
this example shows the change of the sketch from ellipse to circle :
[image]

tested on many versions of freecad and on : 

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

mwlandolsi, 2021-06-30:
could anyone confirm this  bug?

kevenwyld, 2021-06-30:
can you provide an example file?

realthunder, 2021-06-30:
Yes, this is a bug. Thanks for reporting. It's been fixed in realthunder/FreeCAD@a3452b43191411e89deadf6bfb4a7f89a2a86590.

mwlandolsi, 2021-07-01:
Thanks for your emprovement. Have a nice day.

closed 2021-07-01 by mwlandolsi