realthunder/FreeCAD_assembly3 #674: Unable to execute the code because VCRUNTIME140_1.dll was not found !

jpg87Fr, 2021-06-16:
Hello, I unzipped the latest stable version of LinkBranch on my desktop PC and on a laptop that I use infrequently, both running Windows 10.
Problem: on the laptop I received the following message when launching: Unable to execute the code because VCRUNTIME140_1.dll was not found (I had never had a problem before).
I compared the two installations: the file is indeed present in the same places on the 2 computers.
Unfortunate consequence: I have to participate in an online French meeting concerning FreeCAD on Saturday (obviously with my laptop)!
Do you have an idea ?
As a precaution, I uninstalled the official version 0.19 of FreeCAD in case the Windows registry data on FreeCAD disturbs LinkBranch, although this idea is stupid, and as expected it didn't change anything !

realthunder, 2021-06-16:
Please try copy `bin/DLLs/vcruntime140_1.dll` file to `bin` directory and try again.

jpg87Fr, 2021-06-17:
Perfect, everything works.
Thanks again.

jpg87

herrgerd, 2021-06-17:
> 
> 
> Perfect, everything works.
> Thanks again.
> 
> jpg87

If your problem was solved, please always close the issue :)

closed 2021-06-18 by jpg87Fr