realthunder/FreeCAD_assembly3 #676: Bug: Part WB fusion and compound don't respect assigned color

Miksu80, 2021-06-17:
Basically, make a compound or fusion and random its color, it does nothing..

Color set to fusion:
[image]

darker one has aluminium color in it, other doesnt have any assigned color but they're not affected by fusion color:
[image]

When changing sub-parts color, fusion doesn't update with recompute but when trying to assign different color to it.

Coloring information doesn't go to undo stack either and there is no "restore default colors" -option.

I think that these are not link related but anyways they're some sort of bugs..

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

luzpaz, 2021-08-03:
@Miksu80 Can you recreate on latest Daily build + can you attach an example file?

Miksu80, 2021-08-03:
Recreated & attached:

[FusionColor.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6923377/FusionColor.zip)

luzpaz, 2021-08-03:
Thanks. Please always add the full About info ('latest daily' doesn't age well in tickets)

Miksu80, 2022-03-01:
Ok found the issue. When assigning the color to fusion, it doesnt work because the "Map face color" is checked by default:
[image]
So basically if you have previously assigned colors, they dont change -> unchecking this option does the trick and piece will be unicolored. It'd be good to have this option automatically unchecked when doing "random color".