realthunder/FreeCAD_assembly3 #677: Surface creation

parnasist3, 2021-06-18:
OS: openSUSE Leap 15.2 (KDE//usr/share/xsessions/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2

_Regarding surface creation:_

Would it be possible for you to somehow create a way to make a multisection surface with start- plus end-tangency via a couple of common curves/lines? See attached picture.
[image]
It shall start with "Direction vector A2" and be tangent to "Plane A".
Guide curve shall be "Blend curve A1-C1".
It shall end with "Direction vector C2" and be tangent to "Plane C".

[Part_Arbitrary_surface.FCStd.txt](https://github.com/realthunder/FreeCAD_assembly3/files/6675983/Part_Arbitrary_surface.FCStd.txt)
(remove trailing .txt)

parnasist3, 2021-07-16:
Any chance you can have a look at the illusive "Multi-section surface"?

Rabbit-sk, 2021-07-17:
I don't know if that helps you, but here are som links about Curves workbench and Silk workbench:
https://wiki.freecadweb.org/Curves_Workbench/en
https://wiki.freecadweb.org/Curves_GordonSurface
https://forum.freecadweb.org/viewtopic.php?f=3&t=22675

https://github.com/edwardvmills/Silk
https://github.com/edwardvmills/Silk/wiki
https://forum.freecadweb.org/viewtopic.php?f=3&t=60073
http://edwardvmills.github.io/NURBSlib_EVM/
https://github.com/edwardvmills/NURBSlib_EVM/blob/gh-pages/Tutorial%200.01%20ControlPoly4%20and%20CubicCurve4.md
https://github.com/edwardvmills/NURBSlib_EVM/blob/gh-pages/Tutorial%200.02%20ControlGrid44%20and%20CubicSurface44%20-%20page%2001.md
https://github.com/edwardvmills/NURBSlib_EVM/blob/gh-pages/Tutorial%200.03%20Point_onCurve%20ControlPoly4_segment%20ControlPoly6%20and%20CubicCurve6%20-%20page%2001.md
https://forum.freecadweb.org/viewtopic.php?f=22&t=23243

They are not easy no use at first time, but it may be useful for you.

parnasist3, 2021-07-19:
Hi Rabbit-sk, the Silk workbench do get a step on the way to general surface creation. However, it does not handle _the arbitrary ruling multisection surface_ principle which consists of just that, a multisection surface. Edwards version handles dual-/two-section ruling surface, but not multisection.

Curves workbench have recently got an upgrade to handle arbitrary ruling surfaces, but it relays on approximated curves as a base, which is a big no-no CAD-wise.

The "built in" Surface workbench also works with arbitrary ruling surfaces, but it requires the user to choose the ruling surfaces edge as base for resulting surface creation, which is a big no-no topo-naming wise.

So, as it stands right now **it is possible** to make complex surfaces, but it will not survive in a big and complex model.

The "only" thing missing is to make true multisection surface working plus no surface edge selection, it shall be curve selection and the curve shall not be approximated.
However, I get the feeling it takes someone with higher skills to pull this off, hence the request.

parnasist3, 2021-07-24:
If a bold and brave code reader could dive deep in OCC's bruchy code swamp, I'm sure there are hints on how to make an ruled/governed surface according to the first post in this thread.

As an example, the "built in" surface creation tool forces the user to select a previously created surface's edge in order for the algorithm to figure out what surface to use as a ruling/governing/sovereign surface.

This implies the programmer/coder use that surface edge to look up what is its parent.
I suggest to do the other way around, user first choose the curve(i.e. not surface edge) then a ruling/governing/sovereign surface. The algorithm should then check if the curve is a parent to the ruling/governing/sovereign surface.

The ruling/governing/sovereign surface shall ofcourse be made with the multisection tool and use the curve as a guide.

I'm pretty sure that a fill surface must somehow utilize direction vectors at the corner points and "poles"(which basically are end points on a local vector) along the curves.
These "poles" are spread out along each curve in and equidistant way per curve.
So the OCC code do approximate the original curve by dividing it into multiple short curves and placing **one** "pole" at each tiny curve segment start/end.
It seams that, as of today, the OCC algorithm divide the curve in a fixed set of segments/poles (think it is 30?). Now that will pose a problem when the curve is "longer", read: road length.

So the general solution should also take that into account and implement some sort of "pole count" setting along the curve and show it in the 3D-view port.