realthunder/FreeCAD_assembly3 #678: Reproducible crash after right click on spinbox and "clear list"

ttonino, 2021-06-18:
The spinner box gained a right click, which shows a menu with previously used values and a "clear list" option. Selecting "clear list" always leads to a segfault, see below.

Steps to reproduce: New document - News Body - New Sketch - XY plane - creat cicle - close sketch - pad - right click on "length" spinner. A menu is shown. Click on "Clear list" and Freecad exits immediately.

This is with FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210615-glibc2.12-x86_64.AppImage

FreeCAD 0.19, Libs: 2021.615R24301 +3747 (Git)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7fc7ac190210]
#1  0x7fc7ad3c02fc in ParameterGrp::Clear() from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib64/libFreeCADBase.so+0x1c
#2  0x7fc7ae885cef in Gui::PrefQuantitySpinBox::contextMenuEvent(QContextMenuEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib64/libFreeCADGui.so+0xaff
#3  0x7fc7acef29e2 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Widgets.so.5+0x628
#4  0x7fc7acfd057f in QAbstractSpinBox::event(QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Widgets.so.5+0xbd
#5  0x7fc7acec77f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#6  0x7fc7acecc69d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Widgets.so.5+0xe49
#7  0x7fc7ae3c5759 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib64/libFreeCADGui.so+0x89
#8  0x7fc7ac69f606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Core.so.5+0x86
#9  /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Widgets.so.5(+0x1af094) [0x7fc7acf02094]
#10  /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Widgets.so.5(+0x1aff7c) [0x7fc7acf02f7c]
#11  0x7fc7acec77f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#12  0x7fc7acecd31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#13  0x7fc7ae3c5759 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib64/libFreeCADGui.so+0x89
#14  0x7fc7ac69f606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Core.so.5+0x86
#15  0x7fc7aca43ec5 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Gui.so.5+0x677
#16  0x7fc7aca4491f in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Gui.so.5+0x5f
#17  0x7fc7aca31c65 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Gui.so.5+0x63
#18  /tmp/.mount_FreeCAzITP0h/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5b010) [0x7fc7a33c5010]
#19  /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7fc7a8d1c23d]
#20  /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/./libglib-2.0.so.0(+0x554d1) [0x7fc7a8d1c4d1]
#21  /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7fc7a8d1c571]
#22  0x7fc7ac6ce0e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Core.so.5+0x5e
#23  0x7fc7ac69c019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Core.so.5+0x187
#24  0x7fc7ac69fe8f in QCoreApplication::exec() from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib/libQt5Core.so.5+0xfd
#25  0x7fc7ae33ee1e in Gui::Application::runApplication() from /tmp/.mount_FreeCAzITP0h/usr/bin/../lib64/libFreeCADGui.so+0x40e
#26  /tmp/.mount_FreeCAzITP0h/usr/bin/FreeCADLink(+0x44be) [0x560968bdf4be]
#27  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fc7ac1710b3]
#28  /tmp/.mount_FreeCAzITP0h/usr/bin/FreeCADLink(+0x47d5) [0x560968bdf7d5]

realthunder, 2021-06-18:
Thanks for reporting. The problem has been fixed by realthunder/FreeCAD@8650c772fd278c6b1039fe1fccc24e49c994d218

ttonino, 2021-07-16:
Verified, fixed in current daily release. Thanks.

closed 2021-07-16 by ttonino