realthunder/FreeCAD_assembly3 #679: To change selection method to default in Linkbranch 14-06-21 (Stable and daily)

CADrock1, 2021-06-18:
Hello realthunder,

Thanks for the making FreeCAD very useful.

I want to know how to change selection method to default so that all bodies get highlighted on selection. Please see attached screenshots.
[image]
[image]

Here is system infos:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3584 (Git)
Build type: Release
Branch: LinkStage3
Hash: d1e4d1c5abdedda669e54d1a41b5c08077f7e561
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/India (en_IN)

Thanks & regards,
CADrock1

realthunder, 2021-06-18:
I think you accidentally turned on `Show selection bounding box` option as shown below. There many other selection style options grouped in that toolbutton.

[image]

CADrock1, 2021-06-18:
Yes you right, Thanks for quick reply.

closed 2021-06-18 by CADrock1