realthunder/FreeCAD_assembly3 #680: Sketcher/dimension formulas created in Link do not work in 0.20

bambukouk, 2021-06-18:
There is a difference when creating a formulas for dimension in a sketch between Link branch and Main branch, that causes problems when moving between branches.
Is there any particular reason for it? or is it simply an accidental difference?

`OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
`

realthunder, 2021-06-20:
It depends on the content of the expression. The Link branch support extended expression syntax which may not be supported in upstream. Can you post a file which has trouble loading in upstream?

bambukouk, 2021-06-20:
Thank you @realthunder 
I have stripped down the file to leave just the sketch with problematic formulas.

Here is what it looks like in Link branch:

[image]

and the same sketch when opened in FreeCAD 0.20:

[image]

and the file (renamed .zip to permit attaching):

[formula_problem.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6682161/formula_problem.zip)

bambukouk, 2021-06-21:
Thank you @realthunder

closed 2021-06-21 by bambukouk