realthunder/FreeCAD_assembly3 #681: Bug: Skewed fit tolerance after transform

TCatshoek, 2021-06-18:
Hi Realthunder,

First of all, thank you for your amazing work!

I've run in to some strange behavior when using the fit tolerance option in pads and pockets. It appears the new feature gets skewed in a weird way when it is enabled. It appears to align itself to one of the original axes, and only seems to happen when padding from another body that has been rotated using "transform".

https://user-images.githubusercontent.com/5961113/122614513-149e1980-d087-11eb-9a0c-fbaa485d8877.mp4

OS: Arch Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3584 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: d1e4d1c5abdedda669e54d1a41b5c08077f7e561
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

If there is any other info I can provide, I'd be happy to help.

realthunder, 2021-06-20:
Thanks for reporting. The problem has been fixed by realthunder/FreeCAD@cf273a74dd255f7de5f57e062b556c6db80980c5

TCatshoek, 2021-06-20:
Built the LinkDaily branch from source and can confirm it is fixed! Thank you :)

closed 2021-06-20 by TCatshoek