realthunder/FreeCAD_assembly3 #683: Using plane and point from outside Body doesn't work in Body

parnasist3, 2021-06-19:
[image]

**Trying to use a plane and a point from outside of a Body in a Body as basis for a new sketch. It doesn't work.**

To reproduce:
 - make Body active
 - select Plane_D
 - select Point_D_Ruling_corner_D
 - click on "Create new sketch"

It should work by first clicking on "Create new sketch" then selecting point and/or plane.
It shall also work by first selecting point and/or plane, then clicking on "Create new sketch".

[PartDesign_Arbitrary_surface.FCStd.txt](https://github.com/realthunder/FreeCAD_assembly3/files/6680542/PartDesign_Arbitrary_surface.FCStd.txt)

OS: openSUSE Leap 15.2 (KDE//usr/share/xsessions/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2

parnasist3, 2021-06-21:
Super! Thank you so much!!
Now we only need the daily to be updated as well so we can some testing.
But daily seems to be stuck at 20210615?

parnasist3, 2021-07-07:
Any news regarding "Daily" / "Stable" ?

realthunder, 2021-07-08:
Sorry for the delay. Got stuck on some feature. I'll try to make a release this week.

parnasist3, 2021-07-16:
Now it works(in _PartDesign_Arbitrary_surface.FCStd_ attached above), if and only if it's done in this specific order:
 - make Body active
 - start command "Create new sketch" in PartDesign workbench
 - first select plane(Plane_D)
 - then select point(Point_D_Ruling_corner_D)
 - ... and sketch happily ever after!  =O)

Those links created to them plane and point(Import00[3/4]), could they some day be made to not appear in the tree? Or set as a "sub-branch" to be viewed by a plus-sign/arrow in the tree?