realthunder/FreeCAD_assembly3 #685: Unclear how to modify the configuration of an assembly link without affected link target

bgamari, 2021-06-21:
I have been playing around with the configuration table feature (thank you, @realthunder!) and after reading the [FreeCAD forum post](https://forum.freecadweb.org/viewtopic.php?f=17&t=42183&sid=4a30f776a8351ca66cdf290ae54198c6) and [OficineRobotica's great YouTube video](https://www.youtube.com/watch?v=PqARo51D4E0) am still a bit perplexed by the interaction between Link objects and configurations.

My goal is to great a Assembly object for a rod which can be referenced in various length configurations as part of a larger assembly. I have tried the following:

 1. Create a Spreadsheet, using cells `A2` and down for the configuration name and `B2` and down for the length; give cell `A1` an alias of `rod_name`, `B1` an alias of `rod_length`
 1. Create an Assembly for the rod
 1. Within the Assembly create a Body
 1. Within the Body create a Sketch of the rod cross-section
 1. Pad the Sketch
 1. Set the Length of the Pad to `Spreadsheet.rod_length`
 1. In the Spreadsheet, right-client on `A1` and select Configuration Table.
 1. Leave the default "Cell range"; set "Property" to `Assembly.Configuration`. Set the group name to "Rod length" (aside: it would be nice if the label of this field were more suggestive of the fact that it is merely for display; it needn't be a valid Python identifier. Moreover, it would be nice if there were a tooltip describing the "Unsetup" button as its function isn't obvious)

At this point we now have an Assembly with a "Rod length" property, which updates the length of the rod as expected.

However, now try creating a Link to this Assembly. I find that updating the "Rod length" property of the Link also updates the property of the Assembly referenced by the Link. My intuition tells me that the "Copy on change" attribute of the property is relevant here, but I am having trouble understanding the intended semantics of this attribute. What am I doing wrong? Can Links be used in this way?

bgamari, 2021-06-21:
I have attached my test document which I produced using the steps described above. Hopefully this makes it clear what I am after.

[repro.tar.gz](https://github.com/realthunder/FreeCAD_assembly3/files/6687138/repro.tar.gz)

realthunder, 2021-06-21:
The `Link` has a property called `LinkCopyOnChange`. You need to set it to `True` to enable the copy on change feature. You can also use `SubShapeBinder` to do the same thing, which has a `BinderCopyOnChange` property to control the behavior.

bgamari, 2021-06-22:
It seems that setting `LinkCopyOnChange` causes a deep copy of the all dependencies of the referenced object, including the Spreadsheet itself. Is there any way to avoid this?

realthunder, 2021-06-22:
The `Link` copy on change is designed to copy the hierarchy on purpose. The copy stops at the boundary of the current document, meaning that if you have any `Link` inside the hierarchy that links to an external object, the `Link` will be copied, but not the linked object. Alternatively, you can use SubShapeBinder if you do not need the hierarchy at all. SubShapeBinder will also sync changes from the original object.

ceremcem, 2021-06-22:
@realthunder Is "configuration per link" approach supported?

realthunder, 2021-06-23:
Yes. Once 'LinkCopyOnChange' is enabled, the `Link` will make a deep copy if any `CopyOnChange` property published by the linked object changes. Or, maybe you mean something else?

ceremcem, 2021-06-23:
Ah, yes. The problem was that the `LinkCopyOnChange` property is set to `disabled` for every link (in fact this is quite logical and expected) so if we intend to add a new variant by "Link Actions -> Make Link", then we should first set `LinkCopyOnChange` property to `enabled` for the newly created link. 

By the way, what is `owned` value of `LinkCopyOnChange` attribute?

realthunder, 2021-06-23:
`Owned` means the `Link` has been mutated, and it now owns the linked object. Any subsequent changes of the properties marked as `CopyOnChange` won't trigger a copy.

ceremcem, 2021-06-23:
What is the use case of `Owned`? Can you give an example?

And, confused again about the behavior (and/or understood @bgamari's [question](https://github.com/realthunder/FreeCAD_assembly3/issues/685#issuecomment-865435582) now): "Copy on change" concept stands for "copy any changed attribute and leave the rest as a link". 

Example expectation:

* Create `connector` model in Document_A
* Create links named `connector_1` and `connector_2` in Document_B
* Set `CopyOnChange` to `enabled` for both `connector_1` and `connector_2`
* I can set the `TerminalCount` configuration independently for `connector_1` and `connector_2` (Works as expected till here)
* When I made a change in `connector` model in `Document_A`, I would expect that the changes would be reflected to the instances in `Document_B`. 

Now I can't understand why the links wouldn't reflect the changes from the source document.

realthunder, 2021-06-23:
The `Copy on change` concept has two parts. First,  you need to mark some property in the linked object as `CopyOnChange`. Right click the property in the property view, choose `Show all`, and right click again, choose `CopyOnChange`. This process is for the linked object to publish any property for triggering the `Copy on change` behavior. 

When a `Link` is linked to that object, and has its `LinkCopyOnChange` enabled, it will look for any property in the linked object that is marked as `CopyOnChange`, create a mirror of the properties in itself, and monitor for changes. Once the user changed any of the property, the `Link` will perform a deep copy of the linked object, and set its `LinkCopyOnChange` to `Owned`, which prevents the copying if there any further change to those properties.

As described above, the `Owned` state of the `LinkCopyOnChange` is set by `Link` itself. User normally wouldn't do that. After the deep copy is performed, there wouldn't be any linkage between the original and the copied object. Therefore, the change in the original objects won't be reflected in the copies.

The SubShapeBinder has similar behavior to Link, except that it still maintains the link to the original object. And stores the copy in a hidden temporary document. When the user change the `CopyOnChange` property, the binder will discard the copied object and perform a fresh copy.

ceremcem, 2021-06-23:
Okay, now I'm confused even more. There are two "Copy on change" properties, one is `LinkCopyOnChange` (normally visible) and the other is `CopyOnChange` (right click -> show all -> right click). There are so much steps for such a killer feature.

By the way, `CopyOnChange` checkbox is not set on the first click, it is set if you set it "enabled" 2 times: 

[image]

And it's not persistent across some random actions (click on another document and come back, right click, it is unset). 

Let's change our approach to the issue here: What is the "configuration table" intended for? I thought it is there for use cases like creating pin header models, where you may have `N` rows and `M` columns. Then you create one array of pins, make `N` and `M` values (`Number X` and `Number Y`) configurable within the target model by the power of "Configuration Table", so you need to create one configurable model and you can have any number of variants within the same target document by only configuring the links' parameters: 

[image]

The thing is, if the variants within the target document will not track the changes from the source model, this will obviously conflict with the concept of "Parametric CAD Software". 

I think tracking the source model except the changed parameters should be the default behavior. Changed parameters per link should be applied to that link every time the source changes. 

Moreover, the "copy objects into the target document" behavior is frustrating if you want to delete the "Link_object_in_destination", because when you delete the "link" object, all objects within the link is dumped into the document root, so you have to delete every one of them separately. 

> Note: I'm currently designing a commercial device and I'm enthusiastic to benefit from this Configuration Table concept. I'd like to know if this can be achieved within a reasonable amount of time. 

# Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27886 (Git)
Build type: Release
Branch: LinkStage3
Hash: 7437caf7c3e4c6fdcb9f29b6c05596d1268c5248
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2021-06-23:
The `CopyOnChange` in the property context menu is an attribute associated with the property that you right clicked. The `LinkCopyOnChange` is a property of the `Link`. Say you have a `Cube` object you want to make the length configurable. Here are the steps.
1. Right click the `Length` property of the cube object and select `CopyOnChange` to mark the `Cube.Length` property as copy on change.
2. Create a `Link` to the cube
3. Set `Link.LinkCopyOnChange` to enabled
4. Change `Link.Length` to create a variant cube. 

A small detail. At step 2, that is, before `LinkCopyOnChange` is enabled, if you check the property view of `Link`, you'll see a green colored `Length` property, indicating that the property does not belong to the `Link`, but to the linked object, cube. At step 3, after `LinkCopyOnChange` is activated, the `Length` property` turned black, because, `Link` created a property of the same name so that it can be changed without affecting the linked object, and `Link` will monitor the change to know when to copy the cube.

The configuration table is a way to create an enumeration property that can trigger changes in multiple properties of multiple objects at once. So you only need to mark that single enumeration property as `CopyOnChange` to create a configurable design that involves multiple properties in multiple objects.

ceremcem, 2021-06-24:
Will that `Link` track the `Width` of original `Cube.Width` after its length (`Link.Length`) has been modified?

realthunder, 2021-06-24:
No, it will not. For that, you'll have to use SubShapeBinder for now. For a simple single object design, it is straightforward to keep the origin link between them for synchronization. But `LinkCopyOnChange` is intended to be used for object having hierarchies. There is no easy way to determine correspondings between the original and copied objects. It is possible that once the object's configuration is changes, some sub-object will be deleted, and some new ones added. Even if there is no addition and deletion, it is difficult to determine which properties must be synchronized, because a configuration change may trigger many implicit property changes which are considered part of the variations among configurations, and therefore must not be synchronized. 

Of course, it is not impossible to do syncing between Links. For example, I can add option to let user to manual sync, by delete existing copies, and rebuild the variant. But this will be error prone in case you have any other objects that refers to any sub-object in the copied hierarchy.

ceremcem, 2021-06-24:
Is this hard to implement: Anytime we drag an object from `My_Source_Document` and drop into another one there is a `Link` created, lets call this type of link `Link_external`. If you create another link from that `Link`, its type will be `Link_internal`. 

Second and the rest of `Link_external`s will not target to `My_Source_Document` at all, there will be hidden copies of `My_Source_Document` and every single instance of `Link_external` will point a separate hidden document. Whenever we update `My_Source_Document`, those actions will be duplicated on the hidden copies (because the variant information (configuration) is currently kept inside the `Source_Document`). Every time we want to add a new variant, we will drag and drop from the source document. This will remove the necessity of marking the `LinkCopyOnChange` as `Enabled`. This is a workaround.

realthunder, 2021-06-24:
That's awful lots of steps (which I don't fully understand) just to save one extra step of changing `LinkCopyOnChange`. How about I just create another command called `Make variant link` instead?

The sync change behind scene idea is very unlikely to work. Changing the original configurable design has unforeseeable side effects to its depending objects. I really can't see it work in any reliable way, and the best I can offer is a manual operation and the user should prepare for failure and manual fixes. Just imagine how one can reliably sync a sketch change in the original design, where part of the sketch may be fixed by some configuration parameter, hence different among variants.

ceremcem, 2021-06-24:
> That's awful lots of steps (which I don't fully understand) just to save one extra step of changing LinkCopyOnChange

No, it was for preventing the deep copy behavior.

> How about I just create another command called Make variant link instead?

Anything sounds okay if multiple variants would track the changes from the original object. And please let us simply delete any link after changing its configuration. Currently the "**Copy**OnChange" creates a copy, so when we try to delete the "Link", this action deletes the container and puts the contents of container into the root of the document. So we need to delete every remnant object manually.

realthunder, 2021-06-25:
> so when we try to delete the "Link", this action deletes the container and puts the contents of container into the root of the document. So we need to delete every remnant object manually.

That doesn't sound right. Deleting the link does not affect the linked object, so if the linked object is a container, it should still hold  whatever it is holding before. Can you post some screencast showing the problem?

ceremcem, 2021-06-25:
> That doesn't sound right. Deleting the link does not affect the linked object, so if the linked object is a container, it should still hold whatever it is holding before. Can you post some screencast showing the problem?

Correct, the original object was copied into the target document right after making any change on the `Link` while `Link.LinkCopyOnChange` is set to enabled. It was not after deleting  the object, sorry. 

Do the `Make variant link` store a context variable/database per link within the target document?

realthunder, 2021-06-25:
I will add a new mode to `LinkCopyOnChange` to auto rebuild the variant on change in original design. Like I said before, this is an error prone process, especially if you have other objects referring to any sub-object of the copied design. I'll also add a manual rebuild option to the context menu.

ceremcem, 2021-06-26:
@realthunder I'll create the variants with [my old trick](https://github.com/ceremcem/FreeCAD-Asm3-Library/blob/967dc66d7ea0e2b16d417c1847ef7e9cb13bb2e1/tools/update_docname.FCMacro) this time. 

Please let us discuss/understand the inner design of `Make variant link`. This is a killer feature, this should work rock solid. 

For example, I can not understand why it may have errors when the `Link_N`'s data is separately stored within its document and applied everytime the source model is updated. It seems like a rock solid way to go (and this approach works flawlessly in my aeCAD).

realthunder, 2021-06-26:
The information about how to create the variation is stored in the source model, not the variants. When the source model changes, the only way to make sure the variation is faithful is to delete and make a new copy, apply the configuration (i.e. those marked as `CopyOnChange`) and regenerate the variant entirely. That's how `SubShapeBinder` does it.

As to why this can go wrong, it's the same as normal design, where you modify something deep in the history, something may go wrong, and require human intervention. For one thing, the number of properties that is marked as `CopyOnChange` can change, the range of valid value of those properties may change. The geometry of the generated variation is most likely changed, and any objects that depends on the variation and refers to those geometry may break.

ceremcem, 2021-07-09:
@realthunder Can you declare a schedule for addressing this issue? Is there any particular resource required?

realthunder, 2021-07-10:
I am in the middle of a renderer optimization that unfortunately got stuck for several weeks due to its complexity. I do plan to make a release soon, but this link variant improvement may not get there in time. I'll try my best to make it this month.

realthunder, 2021-07-21:
@ceremcem, I just implemented tracking link copy on change behavior. It is in `LinkDaily` branch right now. Can you please build that branch and give it a try first? Or do you prefer to have a separate branch that is based on `LinkStage3` for testing.

There is a new option in `LinkCopyOnChange` called `Tracking` which will auto refresh the copy if the original source object changes. But I would recommend against using this mode. Instead, the normal copy on change still tracks the source object in the background, and will change the tree view icon to indicate the change. Click the icon to manually refresh the copy.

[image]

ceremcem, 2021-07-24:
`LinkStage3` branch is quite appropriate for testing this feature. Current version is: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27928 (Git)
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

However there is no `Tracking` option within `Link Copy On Change` menu in the "instance" properties: 

[image]

Will `Tracking` mode keep the instance specific configuration within its container assembly (thus its container file)?

realthunder, 2021-07-24:
@ceremcem, you can test out the changes in `LinkDev` branch. It is based on `LinkStage3`. This is not really a small change, so better test it first before getting into `LinkStage3`.

There are a few other improvements. The mutated instance is now hidden by default, and will be auto deleted when the link is deleted. You can temporary reveal the mutated instance by using the `Std_LinkSelectLinked` command on the link. If you want to keep the mutated instance when deleting the link, set `LinkCopyOnChange` to `Disabled` before deletion.

ceremcem, 2021-07-24:
Okay, no problem, I'm on it.

ceremcem, 2021-08-23:
@realthunder What is the exact procedure for the following scenario:

1. A configurable model is created (`Model_A`). 
2. Create two instances inside another model (`Model_A_1` and `Model_A_2`). 
3. Choose different configuration for those two instances. 
4. Edit original model (`Model_A`)
5. Expect both instances (`Model_A_1` and `Model_A_2`) follows those changes while still regarding to their unique configuration.

ceremcem, 2021-08-23:
@realthunder Currently setting `Link Copy On Change` to `Enabled` gives following exception: 

```
17:14:19  <Exception> PropertyLinkList does not support external object
...
17:15:34  Recompute failed! Please check report view.
17:15:34  0.002982 <asm3.main> assembly.py(3844): solver exception when auto recompute
Traceback (most recent call last):
  File "<string>", line 533, in _catch
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/solver.py", line 412, in solve
    return _solve(*args,**kargs)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/solver.py", line 378, in _solve
    'Failed to recompute {}'.format(objName(assembly)))
RuntimeError: Failed to recompute (MC-1.5_3-G-3.81#Female) MC_1_5_3_G_3_81#Assembly
```

Insisting changing the new instance's configuration makes FreeCAD crash:

```
Recompute failed! Please check report view.
141.647853 <asm3.gui> gui.py(460): command "asm3CmdSolve" exception
Traceback (most recent call last):
  File "<string>", line 625, in report
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/solver.py", line 412, in solve
    return _solve(*args,**kargs)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/solver.py", line 378, in _solve
    'Failed to recompute {}'.format(objName(assembly)))
RuntimeError: Failed to recompute (MC-1.5_3-G-3.81#Female) MC_1_5_3_G_3_81#Assembly

2.602998 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face6
0.015224 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face5
0.01502 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face2
0.011003 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face3
0.007347 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.Face10
0.007203 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.Face9
0.003824 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face4
0.003056 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_2-G-3.81#Body) MC_1_5_2_G_3_81#Body.Pocket.Face10
0.002552 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_2-G-3.81#Body) MC_1_5_2_G_3_81#Body.Pocket.Face9
Recompute failed! Please check report view.
0.009453 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face6
0.006213 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face5
0.006249 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face2
0.005416 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face3
0.002889 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.Face10
0.002622 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.Face9
0.002714 <asm3.main> assembly.py(741): Failed to get geometry element from (MC-1.5_3-G-3.81#Body) MC_1_5_3_G_3_81#Body.Pocket.?Face4
Recompute failed! Please check report view.
0.002982 <asm3.main> assembly.py(3844): solver exception when auto recompute
Traceback (most recent call last):
  File "<string>", line 533, in _catch
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/solver.py", line 412, in solve
    return _solve(*args,**kargs)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/solver.py", line 378, in _solve
    'Failed to recompute {}'.format(objName(assembly)))
RuntimeError: Failed to recompute (MC-1.5_3-G-3.81#Female) MC_1_5_3_G_3_81#Assembly

Traceback (most recent call last):
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/assembly.py", line 234, in onDocumentRestored
    self.linkSetup(obj)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/assembly.py", line 4045, in linkSetup
    partGroup = self.getPartGroup(True)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/assembly.py", line 4185, in getPartGroup
    self.getElementGroup(True)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/assembly.py", line 4163, in getElementGroup
    self.getConstraintGroup(True)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/assembly.py", line 4127, in getConstraintGroup
    '{}'.format(objName(ret)))
<class 'RuntimeError'>: invalid parent of constraint group main#Constraints
Traceback (most recent call last):
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/assembly.py", line 4668, in finishRestoring
[... 92 lines cut ...]
SoFCUnifiedSelection.cpp(2632): action stack fault
SoFCUnifiedSelection.cpp(2632): action stack fault
SoFCUnifiedSelection.cpp(2632): action stack fault
SoFCUnifiedSelection.cpp(2632): action stack fault
SoFCUnifiedSelection.cpp(2632): action stack fault
SoFCUnifiedSelection.cpp(2632): action stack fault
SoFCUnifiedSelection.cpp(2632): action stack fault
```

# Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28298 (Git)
Build type: Release
Branch: LinkDaily
Hash: aebb0a7ee810a1b7d26a929df6b609af8601fc9b
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2021-08-23:
@ceremcem Can you please send me the file causing the above error? By right, the workflow is just like normal `Link`, with the additional step of the setting `Link Copy On Change` before change the configuration properties.

ceremcem, 2021-08-24:
# Reproduction 

1. Open `main.fcstd`. 
2. Change `main.Assembly.<any part>.Configuration` value from `12 Terminals` to something else. 
3. FreeCAD will crash. 

# Attachment

[test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7036916/test.zip)

realthunder, 2021-08-25:
@ceremcem I've pushed a few fixes to `LinkDev` branch. Please sync and try again.

ceremcem, 2021-08-25:
Crash issue is resolved. Thanks. 

However, I still couldn't manage to update the instances when the source `LIBRARY_MODEL` is modified. Please try changing the red part's color to green. You'll see one of the unmodified instance (the one with 12 terminals) will immediately reflect the change. However, the other one, which is configured to have 2 terminals, will not reflect the change and still display its red part as red.

ceremcem, 2021-08-25:
Ah, I just saw the context menu option: 

[image]

Is there a way to determine if any configurable object needs to be refreshed? Rationale: This process should be automatic. User shouldn't track all the instances within the target model in his/her mind, so we need to find any configurable objects and refresh them `a)` automatically, `b)` interactively by scripting.

realthunder, 2021-08-25:
The Link will track the change in the original configurable object. You can manually recompute the original object and see changes in Link. The problem is that color is a view property and usually do not trigger object recompute. You can use the `Set colors` context menu to change color, which has an option to recompute on finish edit, or just manually recompute it.

ceremcem, 2021-08-25:
Okay, got it. 

Is there any reason for leaving a configurable object's `Link Copy On Change` property as `Disabled`? In other words, shouldn't `Link Copy On Change` set as `Enabled` by default?

ceremcem, 2021-08-25:
There was a one time exception, but I can't remember what I was doing before that. I'm reporting it just in case it means anything to you: 

```
#31  0x7f32e2a02671 in Gui::Application::slotDeleteDocument(App::Document const&) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x131                     [0/1133]
#32  0x7f32e2113c17 in boost::signals2::detail::signal_impl<void (App::Document const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function
<void (App::Document const&)>, boost::function<void (boost::signals2::connection const&, App::Document const&)>, boost::signals2::mutex>::operator()(App::Document const&
) from /home/aea/fc-build/Release/lib/libFreeCADApp.so+0x257
#33  0x7f32e22dcfab in App::Application::closeDocument(char const*) from /home/aea/fc-build/Release/lib/libFreeCADApp.so+0x6b
#34  0x7f32e22dd0dc in App::Application::closeAllDocuments() from /home/aea/fc-build/Release/lib/libFreeCADApp.so+0x2c
#35  0x7f32e2f0e669 in Gui::MainWindow::closeAllDocuments(bool) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x2a9
#36  0x7f32e2ae588d in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x39d
#37  0x7f32df4857f6 in QMetaObject::activate(QObject*, int, int, void**) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x766
#38  0x7f32dfdd0f02 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
#39  0x7f32dfdd3520 in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf0
#40  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2cba4c) [0x7f32dff43a4c]
#41  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2d2f70) [0x7f32dff4af70]
#42  0x7f32dff4bedb in QMenu::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x26b
#43  0x7f32dfe154d8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1d8
#44  0x7f32dff4e36b in QMenu::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x11b
#45  0x7f32dfdd74c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x81                                         
#46  0x7f32dfddebb8 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x458
#47  0x7f32e2a977a8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x68
#48  0x7f32df45c489 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
#49  0x7f32dfdddee9 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /lib/x86_64-linux-gnu/li
bQt5Widgets.so.5+0x1a9
#50  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1b8393) [0x7f32dfe30393]
#51  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1bab5e) [0x7f32dfe32b5e]
#52  0x7f32dfdd74c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x81                                         
#53  0x7f32dfdde970 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x210
#54  0x7f32e2a977a8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x68                                         
#55  0x7f32df45c489 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
#56  0x7f32df808003 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x6e3            
#57  0x7f32df809e85 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x13
5
#58  0x7f32df7e407b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xab             
#59  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0xdd49b) [0x7f32d53e949b]
#60  0x7f32df45b15b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#61  0x7f32df463132 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#62  0x7f32e2a0812b in Gui::Application::runApplication() from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x39b
#63  /home/aea/fc-build/Release/bin/FreeCAD(+0x4880) [0x55c11276c880]
#64  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f32deac709b]
#65  /home/aea/fc-build/Release/bin/FreeCAD(+0x4bda) [0x55c11276cbda]
```

ceremcem, 2021-08-25:
One problem is that, if you change the instance's `LinkCopyOnChange` to `Enabled` but you don't change any configuration of the instance (maybe you are happy with current configuration of the library model?), current configuration is not stored within the instance. This effectively means `LinkCopyOnChange` is `Disabled` because any modification on the library model will be reflected to the instance, which is an undesired behavior.

ceremcem, 2021-08-25:
1. Start with this state, main file is `main`: [test-state-1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7047720/test-state-1.zip)
2. Set those two instances' `Link Copy On Change` to `Enabled` within `main` model.
3. Set one instance's configuration to `4 Terminals`. See there is one 8 Terminal connector and one 4 Terminal connector in main model. 
4. Go to library model (`ffkds_5.08`) and change defauld configuration from `8 Terminals` to `2 Terminals`. 
5. In the main model, there is now one 2 Terminal connector and one 4 Terminal connector. (We don't expect that)
6. There is now a "Refresh" icon appeared next to the 4 Terminal instance. Right click the object (link) and "Rerefresh configurable object". 
7. Whole model is now broken: 

[image]

Resulting project is here: [test-state-2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7047729/test-state-2.zip)

# Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28315 (Git)
Build type: Release
Branch: LinkDaily
Hash: 5483ea00998ff20fe1127b2e1a0d7d91fcfae21a
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

ceremcem, 2021-09-06:
@bgamari I see no problems in my active project when I follow the above instructions. Did the latest changes satisfy the issue?

Here is the complete procedure: 

1. Create a configurable model (`LIB_MODEL`)
2. Mark the configurable model's configuration properties as `Copy On Change`.
    1. Click on the Assembly container of the configurable model. 
    2. In the "Property Window", right click, "Show all"
    3. Go to each configuration property, **left click on it**, then right click, mark the `Copy On Change` property.
3. Drag an drop (create `Link`) `LIB_MODEL` into your target model to create an "instance".
4. Set `Link Copy On Change` property of the instance as `Owned` in your target model. 

Default configuration of the instance will be stored as if it were explicitly set by the user in case of any default configuration changes in `LIB_MODEL`.

bgamari, 2021-11-28:
Indeed, this seems to work quite well. Thank you @realthunder and @ceremcem!

closed 2021-11-28 by bgamari

MatthiasWM, 2022-09-02:
This must be the best hidden description for one of the most essential features of a parametric CAD system. Thanks for taking the time to implement. The missing automatic update is bearable, but the feature itself is grandios!