realthunder/FreeCAD_assembly3 #687: AppImage fails to start (Aborted)

rednaz1337, 2021-06-22:
This is what happens when I try to start the AppImage on Arch Linux
> ./FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210519-glibc2.12-x86_64.AppImage

```
/tmp/.mount_FreeCAHqxqcI
FreeCAD 0.19, Libs: 2021.519R24301 +3554 (Git)
Loading Assembly4 WorkBench
Sheet Metal workbench loaded
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAHqxqcI/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_FreeCAHqxqcI/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAHqxqcI/usr/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCAHqxqcI/AppRun: Zeile 46: 176111 Abgebrochen             (Speicherabzug geschrieben) "$PREFIX/bin/$name" "$@"
```

I think it's the same problem as #660 . But every other OpenGL software works fine. Here's the output of `glxinfo`

```
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel (0x8086)
    Device: Mesa Intel(R) HD Graphics 630 (KBL GT2) (0x591b)
    Version: 21.1.2
    Accelerated: yes
    Video memory: 3072MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
```

However, because my laptop has a second Nvidia GPU, when I launch it using `prime-run`, it starts just fine. Maybe something changed in a recent mesa release? I think it also doesn't work on my other laptop, with an AMD GPU. 

The official FreeCAD 0.19 works without problems.

brychanrobot, 2021-07-03:
https://forum.manjaro.org/t/many-appimages-have-stopped-working-after-todays-update/69445/16 had helpful information and a workaround for this issue.

luzpaz, 2021-07-15:
Closing in favor of #671

closed 2021-07-15 by luzpaz