realthunder/FreeCAD_assembly3 #688: Draft/Array does not play well with assembly (Elements are created with wrong placement)

ceremcem, 2021-06-22:
# Description 

Here is the current state of model: 

[image]

Here is the desired achievement: 

[image]

However, this can't be achieved by Assembly constraints. I guess the problem is that the `Element` placements are wrong: 

[image]

# Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27886 (Git)
Build type: Release
Branch: LinkStage3
Hash: 7437caf7c3e4c6fdcb9f29b6c05596d1268c5248
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```
# Attachments 

[pcb_terminal_block_ffkds_5_08.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6693408/pcb_terminal_block_ffkds_5_08.zip)

realthunder, 2021-06-24:
I can't reproduce this problem in the latest Stable release. Can you please try sync with my asm3 repo and try again?

ceremcem, 2021-06-24:
## Reproduction

1. Create a cube
2. Create an array of cubes using Draft/Array. You'll have 4 cubes by default parameters. 
3. Add an assembly container
4. Move the `Array` into the `Assembly`
5. Click the first (original) cube, click "Create Element". That element will be in the correct location. 
6. Click any face of rest of the cubes, click "Create Element". That element will be placed in the wrong location (it will be placed on the first cube).

realthunder, 2021-06-25:
The fix is in the asm3 repo.

BTW, the file you posted in OP has two problems. First, you are using draft array of an object in a App::Part container, while the array itself is outside of the container. This may cause various weird results. You'll get warning on every recompute to remind you of that. If you really want to do that, the correct way is to use a normal Link to link an object inside the App::Part, and then create Draft array on that link, or, right click the object grouped by the link and choose `Link actions -> Replace with link`.

Second, the Draft has two states, the default collapsed state, where the element objects are not shown in the tree view. In this state, the placement of the elements are dictated by the array, so asm3 will treat the whole array as a single object. Internally, asm3 checks for the `Array.PlacementList` property immutable status. For some reason, this status bit is not properly set in the file you posted. There could be some bug hidden somewhere else. Toggling the `Array,ExpandArray` back and force can fix the missing status bit. Without the bit, asm3 will treat each element in the array as a separate part, cause strange result.

closed 2021-06-25 by ceremcem

ceremcem, 2021-06-25:
Seems fixed, thanks.