realthunder/FreeCAD_assembly3 #692: [Enhancement] Sketcher: Dimensions to be direction-sensitive

Miksu80, 2021-06-24:
When inputting chained dimensions they're fine but if you change a dimension so that it "goes over the second dimension", second dimension is "inverted" (the point-to-point dimension is ok but the line direction changes) and the dimensions are off.

Do a shaft end with snap-ring slot:
[image]
[image]

Now change the shaft length +30mm and notice how the dimensions swap direction:
[image]
[image]

This happens often with parametric tables and everything has to be constrained from center-lines although even this does not always help..

herrgerd, 2021-06-24:
Hm, why would the sketcher do this? I quickly made a part like yours and it works like expected:
[image]

Miksu80, 2021-06-24:
It might have something to do with the order dimensions are put in. So try to input them the opposite order to repeat it. Family keeps me out from the PC for a while..

Miksu80, 2021-06-28:
Here's a file that reproduces it on my pc. Go to sketch and change 30mm -> 50mm..
[image]
[image]

[Sketchtest.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6723266/Sketchtest.zip)

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

herrgerd, 2021-06-28:
Indeed it does switch direction here. Interesting!
I think I had a similar problem not long ago, but can't remember what it was.

@realthunder interesting thing is: It only "breaks" when doing a bigger leap from 30 to 50mm for example. Going up in 5 or 10mm increments does not break the sketch.

realthunder, 2021-06-28:
Other users have reported similar problem. This is a Sketcher solver problem, which I am not that familiar of. I'll take a look when I got time.

Rabbit-sk, 2022-01-16:
Hello RealThunder!

> I'll take a look when I got time.

Is there something new to this issue? I give a small example of this problem. 

_In my opinion, this is the second biggest problem in FreeCAD after the topo naming problem._ 

Fortunately, TNP is solved, but for really reliable models, we also need to solve the orientation of dimensional constraints in the sketches. This problem has already caused me a lot of adjustments when resizing. It happens that sketches in which there are dimensions with respect to the external geometry becomes completely scattered. 

https://user-images.githubusercontent.com/74265320/149664504-2e72a907-0062-4b57-922c-c8b866e701cc.mp4

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Slovak/Slovakia (sk_SK)