realthunder/FreeCAD_assembly3 #693: Support passing a string value as a configuration property

ceremcem, 2021-06-24:
Like "Configuration Table" feature, we might need to configure the instance (`Link`) with a value outside of pre-determined table, like a float number.

A Sigma Profile is the perfect example. You can create the complex model, but you can not know the possible lengths until usage. 

If we could just update a cell of a spreadsheet through one of the `Link`'s configuration properties, we could 
* either set a length through that cell's `Alias`
* or do any complex operation (like passing a complex model number) by parsing it via RegEx.

realthunder, 2021-06-24:
You can mark any number of properties in the linked object as `CopyOnChange`, and a `LinkCopyOnChange` Link will import all of them.

The Spreadsheet offer some convenience functionality to publish an enumeration property to the top level object. You can manually do that as well. Just turn on `Show all`, and add a new property in the top level object, [double bind](https://github.com/realthunder/FreeCAD_assembly3/wiki/Expression-and-Spreadsheet#dbind) it to any property of some child object. Then mark the new property as `CopyOnChange`.

The double bind can work either way. If you intend to store the configuration variable in a child spreadsheet. Then instead of double bind the new property to the cell (which won't work, as cell in a spreadsheet is read only to external object), you can double bind your cell to the top level object's property.

ceremcem, 2021-06-25:
Waiting for "Custom Context per Link" feature to test it out: https://github.com/realthunder/FreeCAD_assembly3/issues/685#issuecomment-867664232