realthunder/FreeCAD_assembly3 #694: Treeview object names does not show correctly

jamestansx, 2021-06-25:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

---
Problem:

The names of objects in the treeview are hidden. This happens to not only a newly created file but also the existing one. 

[image]
[image]

---

realthunder, 2021-06-25:
I can't reproduce the problem. What language do you use for FreeCAD? Can you please post a file with the problem?

jamestansx, 2021-06-25:
I'm using English for my FreeCAD. 
Here is an example [testcad.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6713583/testcad.zip)

herrgerd, 2021-06-25:
> 
> 
> I'm using English for my FreeCAD.
> Here is an example [testcad.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6713583/testcad.zip)
> 
> The same thing happens to the example files in the freecad too.

Works for me...

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

jamestansx, 2021-06-26:
I see. I will re-install it first.

jamestansx, 2021-07-01:
After I reinstalled the FreeCAD, the problem disappears. Closing this issue.

closed 2021-07-01 by jamestansx