realthunder/FreeCAD_assembly3 #695: NameError: name 'translate' is not defined

ceremcem, 2021-06-25:
# Reproduction 

1. Open [the drawing](https://github.com/realthunder/FreeCAD_assembly3/files/6713836/pcb_terminal_block_ffkds_5_08.zip)
2. Select `_Element` and `_Element005` and create a `PlaneAlignment` constraint by using them. 
3. Following exception is thrown: 

```
05:52:19  166.881375 <asm3.main> assembly.py(3844): solver exception when auto recompute
Traceback (most recent call last):
  File "<string>", line 533, in _catch
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/solver.py", line 412, in solve
    return _solve(*args,**kargs)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/solver.py", line 382, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/solver.py", line 87, in __init__
    ret = Constraint.prepare(cstr,self)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/constraint.py", line 518, in prepare
    return mcs.getProxy(obj).prepare(obj,solver)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/constraint.py", line 1265, in prepare
    return super(BaseCascade,cls).prepare(obj,solver)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/constraint.py", line 1254, in prepare
    h = func(*params,group=solver.group)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/system.py", line 369, in addPlaneAlignment
    self.reportRedundancy(count=count, limit=count)
  File "/home/aea/fc-build/Release/Mod/asm3/freecad/asm3/system.py", line 182, in reportRedundancy
    logger.msg(translate('asm3', 'auto relax constraint {} between {} and {}, {}'),
NameError: name 'translate' is not defined
```

# Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27886 (Git)
Build type: Release
Branch: LinkStage3
Hash: 7437caf7c3e4c6fdcb9f29b6c05596d1268c5248
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

Assembly3 is up to date.

ceremcem, 2021-06-25:
Works correctly, thanks. 

@realthunder, You were adding a comment after pushing a hotfix in the past, so we were getting an immediate notification by the e-mail. It was pretty handy.

closed 2021-06-25 by ceremcem

realthunder, 2021-06-25:
I thought you will get a notification when I push a commit that refers to the commit. You don't?

ceremcem, 2021-06-27:
No. I don't know how to subscribe to such a service. It would be pretty handy if Github were sending notifications when any commit is related to that issue, as if the author left a comment to the issue. Is there a way to benefit from such a service?

realthunder, 2021-06-27:
Did some search. Looks like github does not support this. The subscriber only gets notification for new message or state change of the issue. Unfortunately, there is only two state for an issue, open or closed. I used to just close the issue using commit message, but later changed to just referencing.