realthunder/FreeCAD_assembly3 #696: Auto reload documents when they are changed on the disk

ceremcem, 2021-06-26:
As many text editors do, it might be a very useful feature if FreeCAD could reload the document when it's changed on the disk. 

# Use Case 

> Related to https://github.com/realthunder/FreeCAD_assembly3/issues/685#issuecomment-868866886

I can run [a macro](https://github.com/ceremcem/FreeCAD-Asm3-Library/blob/967dc66d7ea0e2b16d417c1847ef7e9cb13bb2e1/tools/update_docname.FCMacro) that sets a cell's value with document's filename. 

This brings a powerful feature: "Create variants by only duplicating the file with a new filename". 

1. Create text file, called `variants.txt`
2. Duplicate the newest document with the contents of `variants.txt`
3. Re-open your project
4. Run that macro. 

That's it. The macro will process all documents recursively and re-replaces the corresponding cells with filename, so, all necessary variables are set within each of the documents. 

If FreeCAD could auto reload the documents on the disk, updating the variants will be a breeze. We will simply fire a script that watches the folder which holds all the variants. When one of the variants changed, the script will replace rest of the files with newest file. If this could trigger a reload, all we need to do to update all variants is simply "Save"ing the modified model.

ceremcem, 2021-06-29:
@realthunder May I require a priority for this and (less importantly #697)? This feature will significantly improve our development.

realthunder, 2021-06-29:
I am not quite sure about your workflow, but since you are already using a script to `replace reset of the files with newest file`, why can't the script just simply close and reopen the relevant documents using FreeCAD API, unless you are using script outside of FreeCAD?

ceremcem, 2021-07-07:
My expectation was simply as follows: 

1. I'll modify my `lib_model`.
2. All variants will be updated (after I answer "yes" to "Reload changed documents [All]" dialog)

In order to achieve that automatic procedure, FreeCAD must track the files for changes and reload them when necessary. 

However, if I add a manual step, like "Update all variants", I could close the relevant document and reopen it. But that will be a manual step.

ceremcem, 2021-07-09:
This approach may help supporting parametric models but can not help supporting parametric models that depend on parametric models. Since it resolves only a limited number of use cases, I'm going to close this issue.

closed 2021-07-09 by ceremcem