realthunder/FreeCAD_assembly3 #698: Using GDML Tessellate Selected Object - Crashes

KeithSloan, 2021-06-27:
I am seeing a regression in 6.15, (5.31 worked just fine).

If I try and use the command/icon 'GDML Tessellate Selected Object' in my workbench, FreeCAD now crashes.
Workbench [https://github.com/KeithSloan/GDML](url) 

Test file 
[testGDMLBox.FCStd.txt](https://github.com/realthunder/FreeCAD_assembly3/files/6721260/testGDMLBox.FCStd.txt)

realthunder, 2021-06-27:
If you are using MacOSX, can you please send me the crash report? The difference in 6.15 is that it is using Python3.9. Also please paste your FreeCAD About information here.

KeithSloan, 2021-06-27:
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

Process:               FreeCAD [20644]
Path:                  /Applications/FreeCAD_Daily_6.15.app/Contents/Resources/bin/FreeCAD
Identifier:            ???
Version:               ??? (2021)
Code Type:             X86-64 (Native)
Parent Process:        ??? [20637]
User ID:               501

Date/Time:             2021-06-27 12:22:40.535 +0100
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Anonymous UUID:        7B9AEDAD-6A4A-444F-F994-68810425664D

Sleep/Wake UUID:       9B614CE4-0987-4610-A41D-94E505B39961

Time Awake Since Boot: 300000 seconds
Time Since Wake:       44000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [20644]

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   libSMDS.dylib                 	0x000000016cb36b7b SMDS_ElementChunk::~SMDS_ElementChunk() + 283
1   libSMDS.dylib                 	0x000000016cb34c49 SMDS_ElementFactory::Clear() + 121
2   libSMDS.dylib                 	0x000000016cb3fd9a SMDS_Mesh::Clear() + 42
3   libSMESHDS.dylib              	0x000000016c5283a6 SMESHDS_Mesh::ClearMesh() + 54
4   libSMESH.dylib                	0x000000016db05c97 SMESH_Mesh::~SMESH_Mesh() + 263
5   libSMESH.dylib                	0x000000016db0619e SMESH_Mesh::~SMESH_Mesh() + 14
6   MeshPart.so                   	0x000000016c489bba MeshPart::Mesher::createMesh() const + 24954
7   MeshPart.so                   	0x000000016c470068 MeshPart::Module::meshFromShape(Py::Tuple const&, Py::Dict const&) + 3624
8   MeshPart.so                   	0x000000016c470f18 Py::ExtensionModule<MeshPart::Module>::invoke_method_keyword(void*, Py::Tuple const&, Py::Dict const&) + 40
9   libFreeCADBase.dylib          	0x000000010dabca0a method_keyword_call_handler + 234
10  libpython3.9.dylib            	0x000000010d085504 cfunction_call + 52
11  libpython3.9.dylib            	0x000000010d0242f8 _PyObject_MakeTpCall + 136
12  libpython3.9.dylib            	0x000000010d159f33 call_function + 323
13  libpython3.9.dylib            	0x000000010d157a7a _PyEval_EvalFrameDefault + 44906
14  libpython3.9.dylib            	0x000000010d0256e2 _PyFunction_Vectorcall + 242
15  libpython3.9.dylib            	0x000000010d028c0f method_vectorcall + 479
16  libFreeCADBase.dylib          	0x000000010dbac806 PP_Run_Method + 342
17  libFreeCADBase.dylib          	0x000000010daf8f41 Base::InterpreterSingleton::runMethodVoid(_object*, char const*) + 49
18  libFreeCADGui.dylib           	0x000000010bf107fc Gui::PythonCommand::activated(int) + 220
19  libFreeCADGui.dylib           	0x000000010bf08a09 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 1129
20  libQt5Core.5.12.9.dylib       	0x000000010e661d64 QMetaObject::activate(QObject*, int, int, void**) + 1800

Thread 1:: com.apple.CFSocket.private
0   libsystem_kernel.dylib        	0x00007fff68a7a0fe __select + 10
1   com.apple.CoreFoundation      	0x00007fff2e8a4c53 __CFSocketManager + 641
2   libsystem_pthread.dylib       	0x00007fff68b35109 _pthread_start + 148
[... 978 lines cut ...]
__UNICODE                          564K        1 
mapped file                       97.9M       48 
shared memory                      640K       15 
===========                     =======  ======= 
TOTAL                              2.9G     2433 
TOTAL, minus reserved VM space     2.4G     2433 

Model: iMac19,1, BootROM 1037.140.50.0.0, 6 processors, 6-Core Intel Core i5, 3.7 GHz, 8 GB, SMC 2.46f13
Graphics: kHW_AMDRadeonPro580XItem, Radeon Pro 580X, spdisplays_pcie_device, 8 GB
Memory Module: BANK 0/ChannelA-DIMM0, 4 GB, DDR4, 2667 MHz, SK Hynix, HMA851S6CJR6N-VK
Memory Module: BANK 2/ChannelB-DIMM0, 4 GB, DDR4, 2667 MHz, SK Hynix, HMA851S6CJR6N-VK
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Feb 28 2020 15:24:56 version 9.30.357.35.32.5.47 FWID 01-9ce4adf3
Bluetooth: Version 7.0.6f7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD ST2000DM001, 2 TB
USB Device: USB 3.1 Bus
USB Device: USB3.1 Hub
USB Device: USB2.0 Hub
USB Device: Lenovo Calliope USB Keyboard
USB Device: USB Optical Mouse
USB Device: FaceTime HD Camera (Built-in)
USB Device: Magic Mouse 2
USB Device: SAMSUNG_Android
Thunderbolt Bus: iMac, Apple Inc., 47.4

realthunder, 2021-06-27:
I think we've hit the same bug [here](https://github.com/trelau/SMESH/issues/37). It seems to only occur on SMESH 9, which is required if build with Python 3.9, while previous build uses SMESH 8.

KeithSloan, 2021-06-27:
Does look likely.

luzpaz, 2022-01-19:
Related https://github.com/trelau/SMESH/issues/37