realthunder/FreeCAD_assembly3 #699: Linear pattern does not hide original shape in offset

mwlandolsi, 2021-06-28:
Hello,
When creating a linear pattern and the create a Transform offset / Position, the transformed shape could not be hidden.

https://user-images.githubusercontent.com/76944804/123623033-e8139b80-d804-11eb-961a-6e3d4e2e3ed5.mp4

mwlandolsi, 2021-07-09:
This happens also when using pattern offset in the dialog window of the linear pattern.

realthunder, 2021-07-15:
I have added this capability. Note that this is not really a bug. The pattern feature is not supposed to change any existing features. But it is possible to remove the last feature in history, in case it is used for patterning. You can also pattern on other feature in deeper history, and it is not possible to transform the first instance of those features.

mwlandolsi, 2021-07-16:
Yes, thank you for your valuable efforts. the daily branch looks OK but in both daily and stable versions, when editing a pattern, the number of occurrences is automatically reset to "1" 

https://user-images.githubusercontent.com/76944804/125984758-0a852fe8-ca13-4440-8399-17d12f0d7571.mp4

mwlandolsi, 2021-07-17:
Yes Great, Looks fine now, but when patterning a BaseFeature, we do not get the same result as for patterning a non-PartDesign feature.

https://user-images.githubusercontent.com/76944804/126044593-be2997cb-68d7-4d32-9fa5-5099c8b9c4d4.mp4

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-07-26:
There are two types of transformation in a pattern feature. One is to transform a sub-feature which is only applicable for a PartDesign Additive/Subtractive/DressUp feature. The other is to transform the whole shape, that is, all feature shape combined, which is also applicable for the base feature imported when you create the body. You can use the `Transform sub-feature` option to control this behavior.

The transform offset change I added can only be applied to a sub-feature transformation that is immediately before the pattern feature. You can however apply transform offset without restriction by using the `New solid` option in pattern feature.

mwlandolsi, 2021-07-26:
Yes, that's it. thank you Zheng Lei. Pleased to get this information.

closed 2021-07-26 by mwlandolsi