realthunder/FreeCAD_assembly3 #700: solid black box instead of pie menu transparency

kevenwyld, 2021-06-28:
I apologize if this ends up being more of a support request, I just can't seem to figure it out.  When I try to use the pie menu a solid black rectangle appears which obstructs the view of whatever was being selected.  I tested and this happens to me with both experimental rendering and "auto".  Please see screenshot below:

[image]

```
OS: Arch Linux (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

kevenwyld, 2021-06-28:
So I found the `Show pie menu as popup` option and unchecked it.  Now things seem to be working as expected so I guess I'm impacted by the "some graphics driver problem" referred to in that option.  Let me know if you'd like me to provide logs or do some testing.  My hardware/graphics configuration is as follows:

```
OS: Arch Linux x86_64
Kernel: 5.12.13-arch1-2 
Resolution: 1920x1080, 1920x1080, 1080x1920, 1080x1920 
  - (I have 4 screens with both horizontal and vertical shared edges)
WM: i3 
Theme: Adwaita-dark [GTK2/3] 
Icons: Adwaita [GTK2/3] 
CPU: Intel i7-5820K (6) @ 3.600GHz 
GPU: NVIDIA GeForce GTX 950 
Memory: 8947MiB / 32030MiB

I have 4 screens with both horizontal and vertical shared edges. 
```

luzpaz, 2021-07-13:
Do you have any other anomalies (in both linkstage3 and in FreeCAD master?)

kevenwyld, 2021-07-15:
I have not noticed anything that's really a specific issue.  The experimental rendering engine does tend to have choppy movement after the application has been open for a while.  For example if I leave a document open overnight and continue work on it the next day.  I know that's not quite what you're looking for though and could be attributed to so many things it's probably not worth chasing down.  A restart of the application clears it up and is probably good practice anyway.  In general everything has been very stable for me using link daily.

I don't use FreeCAD master much but when I test other peoples projects from here on master they seem to work fine.

Is there a specific area that is suspect when considering Nvidia and Linux other than the pie menu that I should test?

kevenwyld, 2021-09-22:
Probably not worth keeping this open.  The workaround has been working just fine.

closed 2021-09-22 by kevenwyld