realthunder/FreeCAD_assembly3 #701: Illegal storage access

Shkolik, 2021-06-29:
Latest few daily versions often produce illegal storage access error... It can be when I trying to save assembly, or close file, or whatever. Can't reproduce it by intention, tho. 

Is there any way to log callstack or at least obtain some information what's going on, so I can work as usual and if error happen I can see and provide this info to you? 

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

bambukouk, 2021-06-29:
Can't recall details, but it had happened to me few times as well, but I am sure, it was always my error...
right now I am on:
`OS: elementary OS 5.1.7 Hera (Pantheon/pantheon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United Kingdom (en_GB)
`
but at the time I was using Win10

Shkolik, 2021-06-29:
> but I am sure, it was always my error...

How it possible? How user error can produce Illegal storage access?

More likely program struggle to save file to the temporary location or something like that. But to debug it it should be some logs what's going on under the hood. Right now I cannot build freecad from sources, so only hope that Realthunder will provide some way to log this issue by using daily build.

bambukouk, 2021-06-29:
I am sorry my memory is so vague.
If it ever happens again, I will report.

In the meantime... what happens when you get error message?
Do you go to the previous version and start again?
or... do you do something to undo the damage?
or does it go away on it's own?

In my case it was always undoing last stupid idea of mine that sorted the problem.
This "stupid idea" usually involved some circular reference in model tree or something like that.

I am not saying that this is the case with you, but I am not clear what happens after you get the error message...

Shkolik, 2021-06-30:
Usually it just said "Save file under new name" and closes. After re-saving file all goes well for some time. But as I said this issue not regular. Today I was working on pretty complex assembly for a 4 hours and no issue, but I pressed ctrl+s pretty often - when this issue happens It's possible to loose big chunk of work... 

About circular reference - some time it happens even when I work on single file without references/links at all.

realthunder, 2021-06-30:
There is no easy way to log call stack in Windows that I know of. FreeCAD has an startup option (`--log-file`) to save its own log to file. But that log probably doesn't help much with this kind of problem. You can try it though. Open a cmd window and cd to the FreeCAD bin directory, run `FreeCAD --log-file <some file>`.

Shkolik, 2021-07-01:
Will try, thank you!

Shkolik, 2021-07-01:
[illegalStorage.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6750538/illegalStorage.zip)

Ok, for now I have 2 examples of this issue. For some reason instead popping dialog with error program just hangs, but error is the same - you can see it in log file. It's weird, but for me nothing significant in log... Maybe it's possible to add some addition logging when this error can occur?

Shkolik, 2021-07-06:
@realthunder , here is another example of this error:

1)	Open FreeCad
2)	Open file 4060CNC.FCStd

[image]

3)	Open and close file ‘my/Ballscrew.FCStd’
4)	Recompute top level assembly in  4060CNC.FCStd
5)	Now link restore and everything looks nice
6)	Click on Close button of FreeCad window (or File/Close all)

[image]

7)	OK button will close dialog and program. 

All files in attachment. 

For me it looks like any unexpected error will produce this dialog or program will hang. 

First of all will be cool to know why this Link cannot be restored at firs time. Maybe it's my fault, but  I don't see any major errors in my workflow... If you can point it will help a lot!

[is-test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6773214/is-test.zip)

PS: Thank you for your great work on this project!

Shkolik, 2021-07-12:
Another observation:

I unchecked partial loading of linked documents and now experiencing error much less. To be honest in last 2 days I saw it only when clicking on Close All document or closing program when there were some document opened. When I tried to close each doc one by one it went fine, but in my case it took a while :)

You can check this behavior on my project: https://github.com/Shkolik/CNC4060

bambukouk, 2021-07-13:
I have now opened and closed @Shkolik  files in Linux and never had any error mesages about illegal storage.
Yet, when I opened it couple of times in Windows I got illegal storage error message every time.

Shkolik, 2021-07-18:
Looks like issue fixed with latest daily. Will test few days, but hope it's it! Thank you!

Shkolik, 2021-08-03:
Everything works fine! No issues with storage access for last 2 weeks.

closed 2021-08-03 by Shkolik

jpoles1, 2021-12-03:
I'm still having this issue all the time on Release 10.15 running on windows, drives me crazy. I wish I had more info, but I really haven't appreciated any clear pattern as to when this happens. I think most often I'm just editing sketches... I'll try and keep an eye out for any patterns to help track down the root of this issue.

Shkolik, 2021-12-03:
Try to disable partial loading of linked files. It may help. Basically the main cause of issue still can be there and, for my understanding it's propagating changes to the linked files. When I working on large project some of the files loose their preview icons for example.

jpoles1, 2022-02-07:
Still having the same isssue unfortunately, even with the latest release. I tried both checking and unchecking "disable partial loading of linked files" without any change. Can try and share some files that are provoking the issue for me if helpful, but it really seems to occur anytime I pull in links from a separate freecad document.

luzpaz, 2022-02-07:
@jpoles1 please always add your full [About](https://wiki.freecad.org/About) info. thanks

jpoles1, 2022-02-07:
Ah, my apologies, this is currently still occuring on:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

But was also occuring previously on:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +3962 (Git)
Build type: Release
Branch: LinkStage3
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

spoons-rchq, 2022-12-18:
For me, uninstalling A2Plus workbench and restarting Freecad 0.20 fixed the illegal storage access problem on Windows 10

luzpaz, 2023-06-26:
@flaviolenz wrong place to report your issue since you aren't using Link branch. Looks like you're using FreeCAD main stable v0.20.2 

Please update to either Realthunder's Link branch (https://github.com/realthunder/FreeCAD/releases/tag/Tip) or update to FreeCAD main branch development version 0.21.x (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)