realthunder/FreeCAD_assembly3 #702: No issue but a extention of the pocket funktion

dingolino, 2021-06-30:
Hi,
is there a better way to send you something - a question a remark?

Just saw a pull request to extend the pocket funktion with "inverse" pocket. I thought it could be usefull. Maybe it would fit into your FreeCAD concept?

https://github.com/FreeCAD/FreeCAD/pull/4828

Best regards,

realthunder, 2021-06-30:
Yes, it is certainly useful. I'll see if I can extended to all features.

herrgerd, 2021-07-01:
That indeed is a nice function to have. Especially for occasions like a hex end on a rod or similar stuff.

czinehuba, 2021-07-28:
Great feature. Although I would use the correct CAD terms such as intersect.
http://www.ann.ugal.ro/im/anale2009/L16%20Article%201%20-%20BRAILA%202009%20Georgeta.pdf

But I would  go one step further. I would get rid of separate additive and subtractive features. And have the the boolean as an option to all. It will save space in the and make it more versatile having Additive, Subtractive, Intersecting features.