realthunder/FreeCAD_assembly3 #703: Using 'model thread' option in hole operation gives error when using a sketch on bottom face

dreetie, 2021-07-01:
**Reproduction**

- make body in part design
- add sketch on xy_plane and make circle 
- pad 25mm
- select bottom face and add new sketch
- make circle in sketch
- use hole operation and select a profile
- check 'Threaded' and 'Model Thread'

When doing the same but instead selecting the Top face of the cylinder everything works as intended.

**Version**
Freecad Link Daily 2021.06.15
[image]
[ThreadFreecadfile.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6746753/ThreadFreecadfile.zip)

herrgerd, 2021-07-16:
@dreetie The problem seems fixed, at least your provided file does work after a recompute. If it does work for you too,
please close this issue :)

dreetie, 2021-07-17:
Problem solved for me as of version 2021.716

closed 2021-07-17 by dreetie