realthunder/FreeCAD_assembly3 #704: [Feature Request] Making the Pick Geometry [G G] function really pick geometry

herrgerd, 2021-07-01:
Hi @realthunder!
This tool might be a core functionality, but I guess addressing the issue here is the best way to show you what I mean.

`G G` is a massively helpful tool, as can be seen in a @oficinerobotica video here:
https://www.youtube.com/watch?v=vUZezMhwwow

But, the real deal would be if it was able to help you sketching! **The issues is very precisely this:** The command doesn't allow you to "use" the selected vertex/line/whatever for further actions other than selecting itself.

Let me bring an example:
[image]
If there's geometry referenced to the root point (or whatever point, really), and you draw another thing on top, it looses its reference when removing the geometry under it.

So, the solution would be super easy: When creating the geometry, just select the root-point with `G G`:
[image]

Unfortunately the command just shows you what's there to select (and lets you select it), but this selection can not be used for anything else.

Using reference-geometry like axis, root point etc is a very basic concept for robust sketches, which don't break or loose reference when deleting geometry. Having `G G` do this would be awesome!

realthunder, 2021-09-17:
@herrgerd Apart from creating constraint, do you think this would be of any use in any of the geometry creation operation?

closed 2021-09-17 by realthunder

herrgerd, 2021-09-17:
Generally speaking at the moment GG does only one thing: It allows you to select one hidden/close to another entity, most of the time this just helps if you want to delete something. Some constraints can't be selected (or found?!) by the command.

Other Suites have the same problem, the tool there (in SolidWorks and Inventor at least) is named "select other" and allows you to select one or _multiple_ pints/lines/constraints.

If `GG` would allow to select more than one entity/geometry it would help sketching a lot. In addition to that, using what is selected by `GG`  as the "second action" would help sketching too.

But, as constraining a sketch is as important as the sketch itself, I think this qualifies enough to be looked into :)

realthunder, 2021-09-17:
The commit referenced above adds support for using `GG` for constraining. The only caveat at the moment is that you'll need to manually hold `Ctrl` key for multi-selection.

herrgerd, 2021-09-17:
> 
> 
> The commit referenced above adds support for using `GG` for constraining. The only caveat at the moment is that you'll need to manually hold `Ctrl` key for multi-selection.

Does not sound like a caveat or even a problem to me. Thank you @realthunder!!!