realthunder/FreeCAD_assembly3 #705: Modeled threads break after changing properties

herrgerd, 2021-07-04:
Hi @realthunder,
[image]

Threaded holes lose all their properties, when you re-enter the properties and change any field there. The dialog looks like this afterwards:
[image]

While you're at it, it would be super sweet if you could stop the re-calculation till the dialog closes. Threads can take a long time to calculate, especially if you need to change a few fields (which you always need to : ) )

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

realthunder, 2021-07-13:
Thanks for the detailed report. The problem is fixed. To stop view update while editing, simply uncheck the `Update view` option. Check again to see the result.

herrgerd, 2021-07-16:
@realthunder works like expected now. THANK YOU!

closed 2021-07-16 by herrgerd