realthunder/FreeCAD_assembly3 #707: Different size of UI elements compared to older Link branch builds

Rabbit-sk, 2021-07-05:
Dear RealThunder, 

on Windows from release version 2021-04-12 the size of UI elements are bigger than expected.

[image]
[image]

Before:
[image]
[image]

Best Regards

herrgerd, 2021-07-06:
This has happened between releases before, quick fix:
[image]

Go there and change the size back to what you're used to :)

YosypC, 2021-07-06:
Hello, as "the constraint" icon dimension I meant inside the sketch on the "viewport", not on the toolbar. Sorry for the misunderstanding 🙂

[grafik]<https://user-images.githubusercontent.com/24494466/124565750-a73e0700-de42-11eb-80a2-b278b12744c2.png>

Rabbit-sk, 2021-07-09:
I wish it was that simple :) 

If you take a closer look at my second and fourth print screen, you'll notice that the `Size of toolbar icons` on both of them is set to `Medium (24px)`. Changing to another size, restarting FC, and then changing back to Medium (24px)` and restarting FC, did not solve it either. Each icon size setting in 2021-04-12 and later releases has a different icon size on the screen than the corresponding icon size setting in 2021-03-28 and earlier releases. 

In addition, it is not only about the size of the icons, but also about the size of the splash screen, the space between menu items (File, Edit, ...), tabs, status bar, navigation cube, in detail the whole UI is affected by some DPI error. Here is a comparison for better illustration: 
[image]
[image]

Rabbit-sk, 2021-07-09:
For example, I've found that the icons and splash screen are 1.25 times bigger.

realthunder, 2021-07-13:
The change came from Qt 5.15 which is first bundled in 04.12 release. To fine tune DPI scaling, please try one of the environment variables described [here](https://doc.qt.io/qt-5/highdpi.html#high-dpi-support-in-qt). You can modify the `RunFreeCAD.bat` file in the release folder.

Rabbit-sk, 2021-07-13:
Fixed by creating the `QT_AUTO_SCREEN_SCALE_FACTOR` enviroment variable set to `1`.

> To fine tune DPI scaling

I think that it is not fine tuning, but fixing.
`Size of toolbar icons` set to `24px` has real size 30px without deailng with `QT_AUTO_SCREEN_SCALE_FACTOR`
vs
`Size of toolbar icons` set to `24px` has real size 24px with `QT_AUTO_SCREEN_SCALE_FACTOR` set to `1`
[image]

While change of the size of the icons may not look so bad, the pixelized appearance of the splash screen certainly won't make a good impression (download and switch between screenshots at full screen to see the difference): 

[image]
vs
[image]

I haven't tried running FreeCAD on another Windows PC, but I think there's a chance that some other users also have this problem and don't know it's a problem or don't know about this solution. 

Wouldn't it be better [to set this variable in the source code](https://stackoverflow.com/a/36057995)?