realthunder/FreeCAD_assembly3 #710: [Feature request] Universal grid for the 3d view.

oficinerobotica, 2021-07-06:
With the latest developments in the display department, the visual assessment of a scene is really starting to fall in to place. I mean it's becoming WOW style visual assessment. 

It would be extremely useful to have a workbench independent configurable grid display. This helps visually to assess the scale and perspective of models displayed in the scene. At the moment we have the highly configurable `Draft Grid` but it has limitations. To have it display a lot of times it needs the DraftWorkbench to be loaded. Also the grid position is not strait foreword to set. 

Ideally a workbench independent  grid would have a behavior similar with the `ground` found in `shadow display`. That is; automatically detecting the "bottom" of the scene and displaying there. Many models are modeled symmetrically to the axis coordinates and having the grid stuck at center is kind of a bummer. 

The grid properties should be displayed in the combo just above the `shadow` settings. In my opinion a good structure would be:
```
**Base**
    |_
    |_
    |_
**Format**
    |_
    |_
    |_
**World Display Settings**
    |_use grid
    |_always show grid
    |_show grid border
    |_use colored axis
    |_main lines every
    |_grid spacing
    |_grid size
    |_grid color and transparency
    |_display at origin/ground
    |_

**Shadow**
    |_
    |_
    |_
```

The `world display settings` could be later populated with other display/rendering options for the scene like `perspective view fov ` `ambient occlusion precision` etc etc. Although `shadow` should be part of world display settings? I'm just trying to reason a coherent structure for the user experience. Surely there are better ideas to structure this.

[image]

herrgerd, 2021-07-06:
A very good idea!
For people like me who turn the grid off everywhere, global grid settings would be beneficial too.

One annoying thing is: The darn grid is automatically turned on when e.g. first switching from the 'Part Design' to the 'Draft' Workbench. Going back to 'Part Design' leaves the grid activated and you have to go back to 'Draft' to switch it off.

oficinerobotica, 2021-07-06:
@herrgerd At the moment as a workaround for the `draft grid` display you can use a global pie entry to toggle on/off the grid. Right click on the top bar > Customize > Toolbars . In the right pane select `Global` from the dropdown > New . Name the Pie `Tools` , uncheck the first checkbox and leave only the `Pie` one . Select the newly created `Tools` entry and assign the shortcut Q then in the left hand pane , in the search box type "grid" and chose `Toggle grid(Draft_ToggleGrid)` . Use the arrows to add the entry to your pie. 

[image]

Add other tools that you frequently use > Profit

https://user-images.githubusercontent.com/26909793/124593058-79ff5200-de5e-11eb-8467-a2911cf93372.mp4