realthunder/FreeCAD_assembly3 #711: BRep_Builder:: Infinite parameter when trying Padding Up to face

YosypC, 2021-07-06:
https://youtu.be/ukO-rCNF_WU

Video and titles are self explanatory. Using Daily I have this problem. Using Stable I have not. W10 2009.

oficinerobotica, 2021-07-07:
@realthunder While trying to clean this specific file I managed to crash FC 2 times. It seems to happen on sketch close after deleting some redundant constraints.  Also, it also seem to happen when a seemingly correct constarint is marked as redundant by the sketcher solver. The crash doesn't happen instantly but on sketch close.
With this file, after the crash, FC will sometimes fail to recover the file.

These are the 2 traces that I got.

```
/tmp/.mount_FreeCAHXYurE
FreeCAD 0.19, Libs: 2021.615R24301 +3747 (Git)
Assembly4 workbench loaded 
flag
flag2
<Exception> File '' does not exist!
<string>(1)<class 'ModuleNotFoundError'>: No module named 'freecad.gears'
<string>(1)<class 'ModuleNotFoundError'>: No module named 'pygears'
<string>(1)<class 'ModuleNotFoundError'>: No module named 'freecad.gears'
PositionBySupport: AttachEngine3D: subshape not found Import001.Face18
<App> GeoFeatureGroupExtension.cpp(577): RotaryKnob#DatumPlane.Support(links to RotaryKnob#Import001) is out of scope:RotaryKnob#Body001 vs. RotaryKnob#Body002
<Exception> Attacher.cpp(842): AttachEngine3D: subshape not found Import001.Face18
Recompute failed! Please check report view.
<App> GeoFeatureGroupExtension.cpp(577): RotaryKnob#DatumPlane.Support(links to RotaryKnob#Import001) is out of scope:RotaryKnob#Body001 vs. RotaryKnob#Body002
<Exception> Attacher.cpp(842): AttachEngine3D: subshape not found Import001.Face18
Recompute failed! Please check report view.
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
<App> GeoFeatureGroupExtension.cpp(577): RotaryKnob#DatumPlane.Support(links to RotaryKnob#Import001) is out of scope:RotaryKnob#Body001 vs. RotaryKnob#Body002
<Exception> Attacher.cpp(842): AttachEngine3D: subshape not found Import001.Face18
Recompute failed! Please check report view.
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
Sketch Sketch002 has partially redundant constraints!
<App> GeoFeatureGroupExtension.cpp(577): RotaryKnob#DatumPlane.Support(links to RotaryKnob#Import001) is out of scope:RotaryKnob#Body001 vs. RotaryKnob#Body002
<Exception> Attacher.cpp(842): AttachEngine3D: subshape not found Import001.Face18
Recompute failed! Please check report view.
Sketch Sketch002 has partially redundant constraints!
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7fa0edc2e210]
#1  /tmp/.mount_FreeCAHXYurE/usr/bin/../lib/libQt5Core.so.5(+0x279580) [0x7fa0ee1fe580]
```
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f47d29b4210]
#1  /lib/x86_64-linux-gnu/libc.so.6(+0x1ebe10) [0x7f47d2b59e10]
```

That being said, in the video, notice how the user attempts to pad to a different body face while  having no active body in the document. At least I don't see it highlighted in the tree. I made the `InnerKnob` body active, created a subshapebinder of the desired face and tried to pad to it but the `BRep_Builder:: Infinite parameter` error persists. There were some problematic sketches, perhaps I need a bit more time to clean the file. Is this a bug or user error?

@YosypC 
 Fusion360 is really forgiving and has allot of great features but the consensus in the community is that FreeCAD is forcing you to adopt solid design principles especially for hard to brake big projects. 
I highly recommend this sketcher tutorial from chrisb.
Forum link: https://forum.freecadweb.org/viewtopic.php?t=30104
Direct pdf link: https://owncloud.gwdg.de/index.php/s/eZisrfTTCCjDEd9/download

At the moment, FreeCAD cannot yet automatically detect closed areas of sketches to be padded. Notice in the screenshot below the ark that I transformed in construction geometry(blue color) . 
[image]

If that ark is represented white, the pad will not work correctly because it interferes with the closed area to be padded. Also, when working on a body make sure you have it active. If you intend to use a master sketch approach from where you use exports or binders like you did it is better to have it outlined at the begginig of you body history tree. In the `InnerKnob` body , I cant understand how you have the first pad from a export of a later sketch. Have you dragged arround features in the tree?

YosypC, 2021-07-07:
My workflow includes having a single master sketch, unfortunately I'm not easily willing to let it go because I like keeping things tidy in one place. I've found workarounds, like exporting child objects of the edges I want to pad, or alternatively select the proper edges before padding. It still requires having trimmed edges but it's something I can do.

The question still stands: why does Pad to face, in this case, work in Stable but not in Daily?