realthunder/FreeCAD_assembly3 #715: [Feature Request] Offset line(s) in Sketcher

YosypC, 2021-07-07:
A function that creates a line (or a selection of lines) which are offset a determined amount from the selection. For now the current way to emulate this with arcs is calculating the radius difference, which is time consuming.

Shkolik, 2021-07-08:
This feature already exists as macro that avalable in addOn Manager and called "SketcherOffset".

[image]
 

Here is a forum thread:
https://forum.freecadweb.org/viewtopic.php?t=49886 

PS: Of course it will be useful to have this feature incorporated in sketcher natively and to work "in a shadow" without creating lots of construction geometry and extra constraints (at least not showing them). But for me it's pretty minor priority and sometimes it's easier to create offset manually than using macro and then clean redundant/wrong constraints.

YosypC, 2021-07-08:
> This feature already exists as macro that avalable in addOn Manager and called "SketcherOffset".

The macro does not apply constraints to the new edges and doesn't work some times, plus it does not create a dimensional, offset constraint, so if you want to change the distance you have to reapply it selecting all the edges, losing time. It doesn't even let you preview the result. 

During my experience with Fusion360 I've used Offset so many times. FC needs it if it wants to compete.

herrgerd, 2021-07-09:
@YosypC also please don't forget that FreeCAD Link Branch is a one-man-show by @realthunder.
It has come a long long way since he's working on it. Be patient...