realthunder/FreeCAD_assembly3 #717: Freecad freezes during startup, save and export (MAC user)

joanperelopez, 2021-07-07:
freecad last version: 2021 6, 15

Freecad freezes several times during startup (beach ball). 
Freecad also freezes when I try to save or export (even the Mac is freezed). It seam something related to box dialog for the name of the file, directory, etc. The same happened with some older versions.

Thanks in advance

thomasgi1, 2021-07-08:
That behaviour is not specific to the asm3 branch. The main FC branch is affected too. The system freeze is caused by some misbehaving security checks performed by MacOS itself, when running unsigned software.
As a workaround you can self-sign your copy of FC as mentioned on the [Known OS specific issues](https://forum.freecadweb.org/viewtopic.php?f=3&t=30573) page of the FC forum. Please follow the instructions of the **Workaround 3** for the 2. MacOS issue **freezing for about a minute on MacOS Mojave (10.14)** on the page.

BR Tom

luzpaz, 2021-08-03:
Closing issue

closed 2021-08-03 by luzpaz