realthunder/FreeCAD_assembly3 #718: [Issue] Symmetry to point doesn't work

YosypC, 2021-07-08:
Trying to put a point in the middle of an edge but the program wouldn't let me. I have tried selecting all the vertices under the spline on a different rectangle which had only the upper spline (so that I managed the fewest possible cases) but it still gives the same error. Am I doing something wrong?
https://user-images.githubusercontent.com/50027752/124955167-23894380-e017-11eb-9bdc-195cbbb77485.mp4

luzpaz, 2021-07-08:
Please get in the habit of always copy/pasting your full [FC About](https://wiki.freecadweb.org/About) info

Rabbit-sk, 2021-07-09:
Delete this constraint:
[image]
here:
[image]
and here:
[image]
before you apply the symmetry constraint.

In this case, the point on line constraint is redundant. Sketcher has sometimes problem to auto remove redundant constraints and needs some help from the user. 

This problem of not removing redundant constraints should be resolved, as the procedure I have advised you is non-intuitive, gained through practice, and this issue is a problem for the new user.

YosypC, 2021-07-09:
> Please get in the habit of always copy/pasting your full [FC About](https://wiki.freecadweb.org/About) info
Understood, original updated.

> Delete this constraint:
> [image]

Didn't work, it tells me to delete other useful constraints, like dimensionals. Plus, when I click "Update" in the sketcher it gives me this error:
`Recompute failed! Please check report view.` 
Thanks for the support!

Rabbit-sk, 2021-07-10:
> Didn't work

I created a new sketch containing only some geometry (point on line, two lines) and the procedure I advised you worked in it. In your case, I think the solver handled it incorrectly for some reason. Something similar has already happened to me with more complex sketches.

herrgerd, 2021-07-11:
[image]
When using the symmetric constraint, the point can't be on the line before. Don't know why but it's always been like that.

Another example I often need:
[image]

YosypC, 2021-07-11:
> When using the symmetric constraint, the point can't be on the line before. Don't know why but it's always been like that.
Not true. I've always done the symmetry while the point sit on the line. However, doing the it the way you pointed out worked in my sketch. Besides, it doesn't make any sense; the point is free to move up and down to "look" for the center, it has no reason to be overconstrained.

I also noted that every time I place a Point, it prompts me the message: Recompute failed! Please check report view.

herrgerd, 2021-07-16:
@YosypC @realthunder seems to work now. Thanks for fixing!

YosypC, 2021-07-19:
It still does not work in my sketch.

closed 2026-01-13 by YosypC