realthunder/FreeCAD_assembly3 #719: [Issue] Traceback (most recent call last):   File "<string>", line 1, in <module> <class 'ValueError'>: Pad is not allowed as external geometry of this sketch

YosypC, 2021-07-08:
[image]
I can't reference edges for the sketch, it gives me this error:

> 23:52:29  Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> <class 'ValueError'>: Pad015 is not allowed as external geometry of this sketch
> 23:52:29  App.getDocument('RotaryKnob').getObject('Sketch009').addExternal(Part.importExternalObject((FreeCAD.getDocument('RotaryKnob').getObject('Part001'),u'Body003.Pad015.;#4de0;:Hb6f,E.Edge21'), (FreeCAD.getDocument('RotaryKnob').getObject('Part001'),u'Body003.Sketch009.'), noSubObject=True),True)
> 23:52:29  Failed to add external geometry: Pad015 is not allowed as external geometry of this sketch

realthunder, 2021-07-08:
Can you please paste here the FreeCAD About information, and also provide a sample file showing the problem.

YosypC, 2021-07-09:
> Can you please paste here the FreeCAD About information, and also provide a sample file showing the problem.
There you go!
[RotaryKnob.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6793527/RotaryKnob.zip)
The edges I want to select belong to "MainFasciaBody", and are the ones around the "LittleUnderTabs" (LittleTabs Body) sketch. Both are under "FrontFascia" part.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Italian/Italy (it_IT)

realthunder, 2021-07-13:
I can't reproduce the problem. I am using the same Windows release image and open the file, edit sketch `LittleUnderTabs`. Add external geometry from Pad015.Edge21, 24. No error printed. Are you sure you didn't miss any steps in your description?

YosypC, 2021-07-13:
I have tried again with the same file I uploaded, this is the result:
```
18:28:02  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Import005 is not allowed as external geometry of this sketch
18:28:02  App.getDocument('RotaryKnob').getObject('Sketch009').addExternal(Part.importExternalObject((FreeCAD.getDocument('RotaryKnob').getObject('Part001'),u'Body003.Pad015.;#187e;:T2927:5:E.Edge21'), (FreeCAD.getDocument('RotaryKnob').getObject('Part001'),u'Body004.Sketch009.'), noSubObject=True),False)
18:28:02  Failed to add external geometry: Import005 is not allowed as external geometry of this sketch
```

Maybe I should try with a clean install? Maybe it's a setting?

luzpaz, 2021-08-21:
@YosypC did you try a clean install?

YosypC, 2021-08-21:
@luzpaz just used the latest Daily, newly installed. It now works.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Italian/Italy (it_IT)

closed 2021-09-14 by YosypC