realthunder/FreeCAD_assembly3 #722: Navigation: middle mouse button does not always center the view

wohltat, 2021-07-09:
The middle mouse button centers the view. But for some unknown reason this fails in around 20% of the cases.
So, it not rarely happens that i have to press the middle mouse button 3 or 4 times in a row to do the action that i want.

This is not critical but a bit annoying. A bit like a small stone in your shoes. You can walk with it but you are happy when it is gone. :)

realthunder, 2021-07-12:
What navigation style are you using?

wohltat, 2021-07-12:
Sorry, forgot to say that.
I use the blender navigation style.

realthunder, 2021-07-12:
The code looks fine. I tried on my computer. I find that for some mouse, like my logitech one, there seems to have some filter built in to ignore brief (i.e. accidental, I presume) middle button press. I need to press firmly with some slight delay to make it register. This doesn't seem to be a problem in my other mouse.

wohltat, 2021-08-17:
Ok, i think i'm pretty sure i know what it is.
If i don't touch the mouse and trigger the middle button externally or press the mouse down firmly so it can't move than the phenomenon disappears. 
If the mouse is not **exactly** at the same position on mouse down and mouse up then it doesn't work.

This is very hard to achieve if you have a sensitive mouse, that moves a pixel while clicking.
A one or two pixel (or custom) radius for clicking would be nice. Maybe combined with a timer so it does not interfere with rotation.

wohltat, 2021-08-18:
I think this is also connected to a similar issue i often see. When i click right button for context menu in tree view, it instantly chooses the first option. When i focus on keeping the mouse still, this doesn't occur.

In both cases there are two uses for the button. One for click and one for holding/dragging. The latter one is very unlikely to be the desired outcome if just a few milliseconds have past, especially if there is only one pixel movement.

Just an idea to implement this:
If we have the time "t" between mouse down and up, and the moved distance "d", we can model the probability for dragging like so
**p = _tanh_ (t/T * d/D)**

T and D are constants for time and displacement. The probability p is 0 if t or d is 0 and increases with t and d and converges to 1.

if probability p > 0.5 then we assume a dragging intention.

We can simplify the equation(inequation) by combining T and D, and we can remove the _tanh_.

**t * d > _arctan_(0.5)*T*D ~= 0.55*T*D = some_magic_constant**

realthunder, 2021-08-19:
At least on my Linux with my Logitech mouse, I am pretty sure that the middle button trigger delay is built in the device, judging by the fact that I didn't install any special driver for the mouse, and the fact that my other non Logitech mouse behaves differently. The software never receives the event notification if the Logitech middle button is pressed briefly.

> I think this is also connected to a similar issue i often see. When i click right button for context menu in tree view, it instantly chooses the first option. When i focus on keeping the mouse still, this doesn't occur.

Are you using Linux? Because the context menu behavior is noticeably different among platforms. On Windows, the menu action can only be triggered by left mouse button release. On Linux, it can be triggered by left or right. So one common Linux user habit  is to hold the right button to bring up the menu and release it for the activation. But you do have a point. It shouldn't trigger if the mouse hasn't moved for at least some distance.

wohltat, 2021-08-20:
yes, im on linux

```
OS: Linux Mint 20.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

> So one common Linux user habit is to hold the right button to bring up the menu and release it for the activation
I'm aware of that. This is what i meant with "holding/dragging" and i think it is a nice way to use the context menu. It just would be good if the other way of using it (via click) wouldn't behave weirdly.
But for me the middle mouse button for centering is more important than the tree view menu behavior, because there is currently no way to move with just one hand in blender style without side effects(#783). Two hand movement via "Shift + middle mouse button" works nicely though.

luzpaz, 2022-06-30:
Is there a way to modify built in mouse behavior to alter this phenomena?